supermerill/SuperSlicer #2847: "cutting corners" -not a bug, but a big thank you!

Spedi-99, 2022-05-29:
### Version
Version 2.3.57.6

### Operating system type + version
Win10 Pro, latest updates

### Behavior
Not really a problem, but more a replay for a function that is mentioned to be given feedback if useful, or otherwise it will be deleted.

In the past days I mounted a 1.4mm CHT nozzle on my VCore3 and tried printing some bigger parts. But I kept getting problems with the first layer.
First layerhight is 0.3mm and the infill didn't connect properly with the perimeter.
The result always looked like this. 
[image]

I tried to set the encroachment to 49% which helped a bit, but it still had some holes here and there.

Today I set the encroachment back to 25%, but I set "cutting-corners" to 100% and look here:
[image]

Awesome!!
Thanks for this feature!
I hope it will stay in the program, because I definitely wanne keep this big nozzle. 
(So fun to watch 1.2mm layers getting poured(!) on the printbed.
 

cheers
Alex

Sirnut, 2022-05-31:
Interesting you have this experience. My understanding was that cutting corners actually decreased the flow, but it seems to have had an effect I didn't think would happen

Was cutting corners really what helped here? I have left mine set to 100% for the longest time but truly don't know the impact its had

Spedi-99, 2022-10-11:
Hi there.I was trying different settings to fill those holes, but nothing worked. But with this setting they where gone. Maybe I was a bit too euphoric. Will try to reproduce it in the next days and keep you informed. CheersAlex