supermerill/SuperSlicer #2858: Wrong infill layer width with "Enforce 100% infill" and wrong infill layer height with Supporting dense layer

KBalage, 2022-06-03:
### What happened?

When using Enforfce 100% infill the slicer (v 2.4.58.2 tested) sometimes change the selected layer width for infill and causing higher volumetric flows than it should. 
[image]

Using "Supporting dense layer" selected layer height almost twice as normal layer height (0.39 instead of 0.2).
[image]

### Project file & How to reproduce

Normal Benchy with Prusa Mini+.

### Version

2.4.58.2

### Operating system

Windows 11

### Printer model

Prusa Mini+

neophyl, 2022-06-03:
For the supporting dense infill part of your issue please read https://github.com/supermerill/SuperSlicer/issues/2839 for an explanation and the settings you need to change. 

As for the other part, when you open a new issue there’s a request for you to attach a project, without the requested saved and zipped project file to work with then it could be anything and is purely guesswork. 

I believe there are also other open issues around the infill topic but I’m not at my pc at the moment so can’t easily run a search for them.

KBalage, 2022-06-03:
Yep, thank you! 
#2839 maybe solves the problem with the "Supporting dense layer" -either "Keep current flow" or "Layer height" seems promising to resolve my problem (flow change cause artifacts on small parts due to this). 

I'll test both, but there's one question mark regarding the "Keep current flow", it seems's it 'undershoo' the current volumetric flow for some reason (project attached).
[image]
[Superslicer_3dbenchy_test.zip](https://github.com/supermerill/SuperSlicer/files/8832066/Superslicer_3dbenchy_test.zip)

Regarding the other problem, "Enforce 100% fill volume", the same project file shows this issue. 

I've found that the "Infill_Perimeters enroachment" to 0% reduce the layer width for these areas, but not fully:

[image]