supermerill/SuperSlicer #2864: Variable Layer Height exit the app

tottiland, 2022-06-04:
SuperSlicer-ubuntu_18.04-2.3.57.12.AppImage

HI

Every time a want to do a Variable Layer Height the app crash and just exit, since the last version, and now on every version I tried. I tried on windows in the same computer and Variable Layer Height works.

Thanks

supermerill, 2022-07-13:
can't reproduce on latest nightly (in a fresh 22.04)

[image]

maybe with a project file?

diggit, 2022-07-17:
I have same issue with 2.4.58.2
- Archlinux x86_64
- Thinkpad T430,  [i7-3610QM with integrated graphics (Intel® HD Graphics 4000)](https://www.intel.com/content/www/us/en/products/sku/64899/intel-core-i73610qm-processor-6m-cache-up-to-3-30-ghz/specifications.html)

Just started SS, then added box primitive and clicked variable layer height option.

Here is backtrace of crashed thread.
(gdb downloded debug symbols on its own. I am not sure if correct ones.)
```c++
Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
exec_node_insert_node_before () at ../mesa-22.1.3/src/compiler/glsl/list.h:180
180	../mesa-22.1.3/src/compiler/glsl/list.h: No such file or directory.
(gdb) bt
#0  exec_node_insert_node_before () at ../mesa-22.1.3/src/compiler/glsl/list.h:180
#1  nir_instr_insert.isra.0 () at ../mesa-22.1.3/src/compiler/nir/nir.c:1155
#2  0x00007fffbead3ba7 in nir_builder_instr_insert () at ../mesa-22.1.3/src/compiler/nir/nir_builder.c:302
#3  nir_get_texture_lod() () at ../mesa-22.1.3/src/compiler/nir/nir_builtin_builder.c:431
#4  0x00007fffbeaec145 in lower_txb_to_txl () at ../mesa-22.1.3/src/compiler/nir/nir_lower_tex.c:820
#5  saturate_src() () at ../mesa-22.1.3/src/compiler/nir/nir_lower_tex.c:842
#6  0x00007fffbeaf3927 in nir_lower_tex_block () at ../mesa-22.1.3/src/compiler/nir/nir_lower_tex.c:1410
#7  nir_lower_tex_impl () at ../mesa-22.1.3/src/compiler/nir/nir_lower_tex.c:1550
#8  nir_lower_tex() () at ../mesa-22.1.3/src/compiler/nir/nir_lower_tex.c:1565
#9  0x00007fffbf6ee2ae in brw_nir_apply_sampler_key.isra.0 () at ../mesa-22.1.3/src/intel/compiler/brw_nir.c:1340
#10 0x00007fffbf420b2d in brw_nir_apply_key() () at ../mesa-22.1.3/src/intel/compiler/brw_nir.c:1399
#11 0x00007fffbf3cceb1 in brw_compile_fs() () at ../mesa-22.1.3/src/intel/compiler/brw_fs.cpp:9835
#12 0x00007fffbf5291df in crocus_compile_fs() () at ../mesa-22.1.3/src/gallium/drivers/crocus/crocus_program.c:1895
#13 0x00007fffbf59f06d in crocus_update_compiled_fs () at ../mesa-22.1.3/src/gallium/drivers/crocus/crocus_program.c:1954
#14 crocus_update_compiled_shaders () at ../mesa-22.1.3/src/gallium/drivers/crocus/crocus_program.c:2497
#15 crocus_draw_vbo () at ../mesa-22.1.3/src/gallium/drivers/crocus/crocus_draw.c:420
#16 crocus_draw_vbo() () at ../mesa-22.1.3/src/gallium/drivers/crocus/crocus_draw.c:356
#17 0x00007fffbebc4006 in tc_call_draw_single() () at ../mesa-22.1.3/src/gallium/auxiliary/util/u_threaded_context.c:3158
#18 0x00007fffbeba3c29 in tc_batch_execute() () at ../mesa-22.1.3/src/gallium/auxiliary/util/u_threaded_context.c:211
#19 0x00007fffbf63e506 in _tc_sync.constprop.0 () at ../mesa-22.1.3/src/gallium/auxiliary/util/u_threaded_context.c:362
#20 0x00007fffbebc0c21 in tc_texture_map() () at ../mesa-22.1.3/src/gallium/auxiliary/util/u_threaded_context.c:2275
#21 0x00007fffbe94b0aa in pipe_texture_map_3d () at ../mesa-22.1.3/src/gallium/auxiliary/util/u_inlines.h:572
#22 st_ReadPixels() () at ../mesa-22.1.3/src/mesa/state_tracker/st_cb_readpixels.c:530
#23 0x00007fffbe90f517 in read_pixels () at ../mesa-22.1.3/src/mesa/main/readpix.c:1178
#24 _mesa_ReadnPixelsARB() () at ../mesa-22.1.3/src/mesa/main/readpix.c:1195
#25 0x00007fffbe90f846 in _mesa_ReadPixels() () at ../mesa-22.1.3/src/mesa/main/readpix.c:1210
#26 0x000055555617d43b in Slic3r::GUI::GLCanvas3D::_mouse_to_3d(Slic3r::Point const&, float*) ()
#27 0x0000555556191837 in Slic3r::GUI::GLCanvas3D::render() ()
#28 0x0000555556193424 in Slic3r::GUI::GLCanvas3D::_refresh_if_shown_on_screen() ()
#29 0x00005555561935b8 in Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&) ()
#30 0x0000555556fc92ee in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
#31 0x0000555556fc96c4 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
#32 0x0000555556fc984f in wxEvtHandler::TryHereOnly(wxEvent&) ()
#33 0x0000555556fc98ea in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
#34 0x0000555556fc996d in wxEvtHandler::ProcessEvent(wxEvent&) ()
#35 0x0000555556fca327 in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
#36 0x0000555556d96060 in wxWindowBase::SendIdleEvents(wxIdleEvent&) ()
#37 0x0000555556d96038 in wxWindowBase::SendIdleEvents(wxIdleEvent&) ()
#38 0x0000555556d96038 in wxWindowBase::SendIdleEvents(wxIdleEvent&) ()
#39 0x0000555556d96038 in wxWindowBase::SendIdleEvents(wxIdleEvent&) ()
#40 0x0000555556e5d33f in wxFrame::SendIdleEvents(wxIdleEvent&) ()
#41 0x0000555556ca95ad in wxAppBase::ProcessIdle() ()
#42 0x0000555556df6d41 in wxApp::DoIdle() ()
#43 0x0000555556df6e13 in wxapp_idle_callback ()
#44 0x00007ffff6d9cc6b in g_main_dispatch (context=0x5555580e41d0) at ../glib/glib/gmain.c:3417
#45 g_main_context_dispatch (context=0x5555580e41d0) at ../glib/glib/gmain.c:4135
#46 0x00007ffff6df3001 in g_main_context_iterate.constprop.0 (context=0x5555580e41d0, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/glib/gmain.c:4211
#47 0x00007ffff6d9c1cf in g_main_loop_run (loop=0x55555b0ec930) at ../glib/glib/gmain.c:4411
#48 0x00007ffff7564e2f in gtk_main () at ../gtk/gtk/gtkmain.c:1329
#49 0x0000555556e05fc5 in wxGUIEventLoop::DoRun() ()
#50 0x0000555556f1bacd in wxEventLoopBase::Run() ()
#51 0x0000555556ee71ed in wxAppConsoleBase::OnRun() ()
#52 0x0000555556f6ab12 in wxEntry(int&, wchar_t**) ()
#53 0x0000555555dfa982 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#54 0x000055555577a5a8 in Slic3r::CLI::run(int, char**) ()
#55 0x0000555555730c9a in main ()
```

aleung, 2022-09-19:
2.5.59 on Ubuntu 22.04
Crash every time when click the variable layer height button

Project file attached: [crash.zip](https://github.com/supermerill/SuperSlicer/files/9600039/crash.zip)

supermerill, 2024-05-21:
Can't reproduce on my linux vm.
Seems like your opengl drivers crash when calling `glReadPixels`
Seems like [intel opengl is buggy](https://stackoverflow.com/questions/68997497/glreadpixels-causes-crash)
I'll try to do something, but I can't test rn.

closed 2024-06-12 by supermerill