supermerill/SuperSlicer #2866: G1 F-nan(ind) in gcode

filotto78, 2022-06-05:
**Bug description**
SuperSlicer generated a gcode with G1 F-nan(ind) instruction.

**Expected behavior**
Don't see this instruction on the generated gcode.

!!>> Project File <<!!
In the zip file you will find only the gcode since 3mf file zip was too big.

[gauntlet_cover gcode.zip](https://github.com/supermerill/SuperSlicer/files/8840368/gauntlet_cover.gcode.zip)

The emf file can be downloaded here:
[https://drive.google.com/file/d/1wGfEeGs8THRr6s8CxewHMWvDQ3GtUQSF/view?usp=sharing]

**To Reproduce**
Steps to reproduce the behavior:
1. Open the 3mf
2. Slice the model (this is a multimaterial model)
3. Export the gcode
4. See error on gcode
[gauntlet_cover gcode.zip](https://github.com/supermerill/SuperSlicer/files/8840368/gauntlet_cover.gcode.zip)

**Screenshots**
[image]

[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 2.3.57.12
 - Voron V2.4

**Additional context**
This is a multimaterial file.

filotto78, 2022-06-06:
Access to the file stored in Google Drive has been granted.

Slabity, 2022-06-23:
I am seeing this as well, and for me it appears to be related to either the length of time of the print or the number of objects I am trying to print.

[Here is the STL](https://drive.google.com/file/d/1Qy8PRU5yJr2yCEsQ_XIAIeQqWhnDvQ34/view?usp=sharing)

[Here is the 3mf](https://drive.google.com/file/d/1Mupj5t1961WPptJOgZRGil4oZZYT2X05/view?usp=sharing)

I am able to place 5 of those objects in a single print, but the moment I add a sixth one it seems to screw up the processing and show the following for times:

[image]

Not sure if it matters, but I am using Linux in case it is specific to that platform.

marcuswu, 2022-06-28:
I am experiencing something similar that I believe is related.

Instead of nan, I'm getting inf for used filament:
[image]

The gcode has a `G1 F0` which the printer doesn't like and inf extruder movements.
[image]

Note that the print begins with the purge / wipe tower and the first layer of that prints successfully before the printer encounters the problem gcode. The relevant files are linked below:

[ball halves.3mf](https://drive.google.com/file/d/1YSFw_tUb4etQe4FJlmhOIFDZPB6SEssj/view?usp=sharing)
[ball halves.gcode](https://drive.google.com/file/d/1GV1dAEKCbPte1fS0f3BGHixN7G4gYQDg/view?usp=sharing)
[ball half 1 stl](https://drive.google.com/file/d/1Rl8dn7W3lh0w3Qbf_onXaHVsSAEAX42J/view?usp=sharing)
[ball half 2 stl](https://drive.google.com/file/d/1lymPC6Tu_YAJXWvzDzUXhJK-zVjsRqwg/view?usp=sharing)

supermerill, 2022-07-12:
can't replicate on latest nithgly. Should be fixed in next release

Slabity, 2022-08-26:
I can confirm this has been fixed on my end (using 2.4.58.3). No longer running into the same issue.

If nobody else is having this problem then I think it's safe to close.

closed 2024-02-11 by supermerill