supermerill/SuperSlicer #2871: X Window System error on Archlinux

schulzh, 2022-06-07:
### What happened?

After I've upgraded my System yesterday, Superslicer won't start anymore. I've tried the following AppImage versions: `2.3.56.9`, `2.3.57.12`, `gtk2-2.3.57.12` and `2.4.58.2`. All fail with the same error:
```
./SuperSlicer-ubuntu_18.04-2.4.58.2.AppImage
[2022-06-07 16:23:46.330267] [0x00007f82d5915480] [trace]   Initializing StaticPrintConfigs

(superslicer:2005816): Gdk-WARNING **: 16:23:46.918: The program 'superslicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 319 error_code 2 request_code 152 (GLX) minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

### Project file & How to reproduce

N/A

### Version

2.3.56.9

### Operating system

Archlinux

### Printer model

N/A

Wright4TheJob, 2022-06-27:
I am getting the same error on Pop!_OS 22.04 (Ubuntu based): 
> ./SuperSlicer-ubuntu_18.04-2.3.57.12.AppImage 

(superslicer:130713): GLib-GObject-WARNING **: 15:57:17.818: invalid (NULL) pointer instance
(superslicer:130713): GLib-GObject-CRITICAL **: 15:57:17.818: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(superslicer:130713): GLib-GObject-WARNING **: 15:57:17.818: invalid (NULL) pointer instance
(superslicer:130713): GLib-GObject-CRITICAL **: 15:57:17.818: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(superslicer:130713): GLib-GObject-WARNING **: 15:57:17.818: invalid (NULL) pointer instance
(superslicer:130713): GLib-GObject-CRITICAL **: 15:57:17.818: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(superslicer:130713): GLib-GObject-WARNING **: 15:57:17.818: invalid (NULL) pointer instance
(superslicer:130713): GLib-GObject-CRITICAL **: 15:57:17.818: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(superslicer:130713): Gdk-ERROR **: 15:57:17.867: The program 'superslicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 306 error_code 2 request_code 152 (GLX) minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

System packages are updated:
libgtk-3-0 is already the newest version (3.24.33-1ubuntu2).
libgtk-4-1 is already the newest version (4.6.3+ds1-0ubuntu1).

rufinus, 2022-08-09:
Confirmed on Ubuntu 22.04

Seems to be a upstream problem - as I have the same error on latest stable PrusaSlicer.

wannabecoffeenerd, 2022-12-04:
Having the same issue on Ubuntu 20.04 as well.

nighi, 2024-04-21:
Same error after updating to the newest SuperSlicer_2.5.59.8_2024-02-14 on Manjaro (Arch based).
```
[2024-04-21 18:33:49.994608] [0x00007f7d20c17800] [trace]   Initializing StaticPrintConfigs

(superslicer:468664): Gdk-WARNING **: 18:33:51.112: The program 'superslicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.

```

nighi, 2024-04-21:
I've tried to run it with GDK_SYNCHRONIZE in gdb, but not sure if I did it right and superslicer actually started:
``` gdb
(gdb) exec-file superslicer
(gdb) r
Starting program: /usr/bin/superslicer superslicer

This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.archlinux.org>
Enable debuginfod for this session? (y or [n]) y
Debuginfod has been enabled.
To make this setting permanent, add 'set debuginfod enabled on' to .gdbinit.
warning: Probes-based dynamic linker interface failed.                                                         
Reverting to original interface.
[Detaching after fork from child process 471701]
[New LWP 471702]
[New LWP 471703]
[LWP 471703 exited]
[LWP 471702 exited]
process 471688 is executing new program: /usr/lib/appimagelauncher/binfmt-bypass
[Thread debugging using libthread_db enabled]                                                                  
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[Detaching after fork from child process 471707]                                                               
[2024-04-21 18:50:07.113866] [0x00007ffff578e800] [trace]   Initializing StaticPrintConfigs
No such file: superslicer
[Inferior 1 (process 471688) exited with code 01]
```