supermerill/SuperSlicer #2874: Feature "Make Overhangs Printable" like in Cura

Dominik-1980, 2022-06-09:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
Add the possibility to "make overhangs printable" like the option in Cura.

Would be great if this will be implemented. This is the only thing I am missing in SuperSlicer.
But I'm missing very much! It is a game changer.

inspectionsbybob, 2022-09-25:
I will second the request.

theschles, 2022-10-19:
Duplicate of #3201 -- please close this or the other one?

inspectionsbybob, 2022-10-20:
I closed my Duplicate, just making sure this one is still open.

odie5533, 2023-06-27:
Would really appreciate having this option!

Cura has this feature in their CuraEngine. They configure the setting as conical_overhang_enabled (which they show in the interface as "Make Overhangs Printable") https://github.com/Ultimaker/CuraEngine/blob/main/src/FffPolygonGenerator.cpp#L233

And the actual ConicalOverhang function is in https://github.com/Ultimaker/CuraEngine/blob/main/src/ConicalOverhang.cpp

kozross, 2024-03-24:
There's a similar feature in OrcaSlicer as well, which might make it easier to port.