supermerill/SuperSlicer #2880: Exclude Objects not working with Superslicer 2.4.58.2

bob92553, 2022-06-13:
### What happened?

i set up exclude objects last night and it would not activate
i fallowed the instructions
enabled the Label objects setting in your slicer

Enabled the G-Code preprocessor in Moonraker

Enabled exclude_object module in Klipper

sliced new set of files. 
 uploaded and hit print and get this error message in the console

02:45:10
Unknown command:"DEFINE_OBJECT"
02:45:10
Unknown command:"DEFINE_OBJECT"
02:45:10
Unknown command:"DEFINE_OBJECT"

when i went back to SuperSlicer_2.3.57.12 problem went away. and works as expected. 

### Project file & How to reproduce

[sliced gcode.zip](https://github.com/supermerill/SuperSlicer/files/8890521/sliced.gcode.zip)

### Version

2.4.58.2

### Operating system

windows 10

### Printer model

voron trident 350

ayufan, 2022-06-16:
Update Moonraker, and Klipper. The commands used by Klipper are `EXCLUDE_OBJECTS_DEFINE`.

bob92553, 2022-06-20:
uninstalled SS 2.4.58.2 went back to previous version and then installed 2.4.58.2 and now its working. not sure what i did.

closed 2022-06-20 by bob92553