supermerill/SuperSlicer #2884: Build plate on Prusa mini+ turns black when cursor enters 3d view screen

SylvesterMoltisanti, 2022-06-18:
### Version
_Version of SuperSlicer used goes here (help->about)_2.4.58.2

### Operating system type + version
_What OS are you using, and state any version #s_windows 11

### Behavior
* _Describe the problem_Build plate on Prusa mini+ turns black when cursor enters 3d view screen-fine before that-all other buildplates are fine visually-minor color fluctuations on tip of the day
* _Steps needed to reproduce the problem_just turn on and move cursor over work area
* _Expected Results_maintain prusa logo and platform coloring
* _Actual Results_black square with grey border
  * _Screenshots if useful_
[image]

_Is this a new feature request?_yes
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here
[black platter.zip](https://github.com/supermerill/SuperSlicer/files/8933175/black.platter.zip)
_

Wovchik, 2022-08-27:
Have same issue only one Printerpreset works. some other Presets have the same texture as the working one.

JanB97, 2023-11-16:
I just got a new AMD GPU (7800xt) and have this problem with it.
It seems to be fixed in Prusaslicer 2.5, is there any developement on this for superslicer?

JanB97, 2023-11-16:
> I just got a new AMD GPU (7800xt) and have this problem with it. It seems to be fixed in Prusaslicer 2.5, is there any developement on this for superslicer?

This seems to be the commit that fixed it in Prusaslicer, maybe it helps: 
[https://github.com/prusa3d/PrusaSlicer/commit/9430387354078c9e36508824ebd83e683183f753](https://github.com/prusa3d/PrusaSlicer/commit/9430387354078c9e36508824ebd83e683183f753)