supermerill/SuperSlicer #2890: ODD-EVEN layer settings extension to improve adhesion and structural strength

jeedee23, 2022-06-21:
### Version
2.4.58.2

### Operating system type + version
Win 11 Version	10.0.22000 Build 22000

### Behavior
Sometimes perimeters in small objects like these, are not anchored completely to the infill, or the infill pushes the perimeter outwards.
_Is this a new feature request?_YES
I use two identical gcodes, with 2 different settings and merge the two into one: 
Settings for even layers are infill_before_perimeters TRUE, and perimeter width 140
Settings for odd layers are  infill_before_perimeters FALSE, and perimeter width 110
Both have extra_perimeter_on_odd_layers TRUE

The attached files contain:
The 'even' version project
The 'odd' version project
The 'even' gcode
The 'odd' gcode
The 'merged' gcode

Already submitted this once, without a response. A feature 'infill_before_perimeter' TRUE with optional 'infill_before_perimeter_only-on_even_layers' TRUE or FALSE would give a great advantage.

A second feature: perimeter_width_even and perimeter_width_odd would be appreciated. 
[2022-06-21(at 23h08m47s)_new feature.zip](https://github.com/supermerill/SuperSlicer/files/8952877/2022-06-21.at.23h08m47s._new.feature.zip)

karlla1220, 2022-06-29:
I thought it is already supported in the setting. Is this different from your project?

[image]

jeedee23, 2022-06-29:
Hi, this is '**extra-perimeters-odd-layers**', so different from my project.

existing parameters are :
**perimeter-extrusion-width** (and auto calculated perimeter-extrusion-spacing)
**external-perimeter-extrusion-width** (and auto calculated perimeter-extrusion-spacing)

The **extra-perimeters-odd-layers** is already very good, but **perimeter-extrusion-width-even and perimeter-extrusion-width-odd** would be an even better solution. 
Since the code already contains a subroutine that handles the odd layers, I thought it would be a good idea to implement this. 
Now perimeters are built one upon another in a straight vertical line (like you would build a brick house with the bricks aligned, not with a masonry bond). If one changed the perimeter width on odd-even, you have a superstrong masonry-like bond. Did it, and now use it all the time with my own script (on top of that I also use extra-perimeters-odd-layers)
It is very difficult to have a perfect adhesion for the perimeters, without overextruding, this way, one can slightly underextrude and still have a good adhesion because perimeters are 'zig-zag' one on top of another.