supermerill/SuperSlicer #2895: Bad Top Layer, only SuperSlicer

Wiikendz, 2022-06-25:
Suddenly the top layers printed with superslicer look very bad, they look overextruded / the nozzle digs into it creating a rough surface to look and feel. 

There was no issues just a few days ago and settings havent changed, tried both of the latest release without any difference. Printing the same part in Cura with the same slicer and printer settings gives a result like what superslicer gave a few days ago.

This image shows the issue, however mine is not as extreme, in this case left printed with PrusaSlicer and right SuperSlicer
[image]

Down below the difference for me is shown between SuperSlicer and Cura, the Cura being all smooth while SuperSlicer creates a bad top layer. The Cura layer being how SuperSlicer top layer used to be.
[image]

[image]

I cant figure out whats causing this and where it came from, has to do with slicer since others work fine.
Ender 3 S1 Printer with Klipper v0.10.0-481-g0256967d and Mainsail v2.2.1
SuperSlicer 2.4.58.2
Windows 10 (build 19044), 64-bit editio
[issue.zip](https://github.com/supermerill/SuperSlicer/files/8985373/issue.zip)

matthewlloyd, 2022-06-25:
The filament extrusion multiplier in your G-code file is set to 1. Slic3r derived slicers including SuperSlicer need to use an EM closer to 0.95 to avoid overextrusion. This is related to the model used to represent the volume of filament needed. It's a well known issue dating back several years:

https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-others-archive/updated-slic3r-pe-over-extrusion-and-cooling-solved/

This can also be achieved by setting a flow rate of 95%, as some stock PrusaSlicer profiles do.

Wiikendz, 2022-06-26:
> The filament extrusion multiplier in your G-code file is set to 1. Slic3r derived slicers including SuperSlicer need to use an EM closer to 0.95 to avoid overextrusion. This is related to the model [...]

Even though according to calibration mt flow should be at 1?

Kaelum, 2023-05-06:
This has apparently gotten worse with the 2.4.58.5 release.  SuperSlicer is over extruding by ~25% now.  I was wondering why my Rat Rig was over extruding excessively, where I am getting perfect prints when I drop the extrusion to 80% in Klipper.  To verify the issue, I attempted to print the exact same model on my Ender 5 Plus, only changing the printer, and had the exact same results.  The over extrusion is now significantly worse than what you are showing in your pictures.

This can be replicated very easily by printing a solid object and setting the infill to 100%.  You will see just how bad it is after about 6 layers.  Eventually the head will embed itself in the print, and the print will dislodge from the bed. This is a serious bug, and needs to be fixed.  I'm going to run some tests on PrusaSlicer, but I have never been required to use anything other than 1 on a perfectly calibrated printer.

skuep, 2023-09-03:
Interesting thread to find, after weeks of hardware troubleshooting. I just found out it's a superslicer issue after printing the model with Cura and perfect surfaces. 

What's going on? Why are there not more people having this issue? Is it a specific set of settings producing this?

Kaelum, 2023-09-03:
@skuep the issue actually originates in PrusaSlicer, as the exact same thing is occurring with it and SuperSlicer is a fork of PrusaSlicer.

skuep, 2023-09-04:
Oh, okay. In the above comments it sounded like PrusaSlicer didn't have any issues. But maybe it's a version thing.

skuep, 2023-09-08:
I tested orcaslicer, which also works flawlessly. It also fixes underextrusion issues I had in the *bottom* solid infill (when going above 100% line width). Didn't think that was related.

Man, I don't know. I have a feeling I could solve this by nuking my superslicer profiles and restart from scratch. To be honest I kind of like orcaslicer. Somehow I don't take as long to look for specific settings 😅 and I have yet to find a feature that is missing. Sad to see Superslicer go for now.

Kaelum, 2023-09-08:
I have a similar conundrum:

- I can't go back to PrusaSlicer as it has removed several needed options from the UI.
- Bambu has not been around long enough for me to trust OrcaSlicer.
- I fear that SuperSlicer's days are numbered because of the above.

supermerill, 2023-11-22:
What can i do?

Kaelum, 2023-11-22:
@supermerill I don't believe that there is anything for you to do, other than possibly making it easier for people to calibrate filaments.  If you have your E-steps perfectly calibrated, that does not mean that you won't have over or under extrusion, as each filament has different characteristics.  Almost all filaments expand when heated, and remain in an expanded state even after cooling.  Every single `Filament Setting` needs to have its `extrusion multiplier` properly calibrated.  The `Shrinkage` setting also needs to be calibrated, which is somewhat tricky if people don't understand the difference between it and the `Extrusion multiplier`.

If the values are perfectly set, then you will achieve a perfectly flat top even when using 100% or solid infill.  To calibrate it my self, I start with a flat calibration cube (no X, Y, or Z) and print it with solid infill.  I usually start at 0.98 for PLA, and 0.895 for ABS & ASA.  If I see under or over extrusion, I adjust the `Extrusion factor` in the Klipper UI and continue to watch it printing to see if I've adjusted it enough.  In the end, once I have at least 10 consistent layers, I calculate a new `Extrusion multiplier`:

`New Extrusion multiplier = <previous Extrusion multiplier> x <Extrusion factor>`

kingofmonkeys, 2024-05-10:
Just adding to this.  Started using SuperSlicer with my Voron and I've been fighting what looks like over extrusion on solid infill and top infill.  Sounds like others are having the same issue, I'm working my way down with the extrusion width and flow settings trying to get good prints, just wanted to add to the group saying it does look like an issue.  I haven't tried another slicer on this Voron so I can't say for sure Cura wouldn't have the same issue.

supermerill, 2024-05-15:
in next beta, there is a new flow&overlap&speed calibration, using the weight of samples. should hemp with that.