supermerill/SuperSlicer #2902: Above bridge speed adjustment

Ryan6338, 2022-06-29:
When using low layer heights, the layer above bridges needs to be printed much slower than regular infill to achieve good adhesion as the thin bridge layer just warps under the nozzle pressure rather than fusing. It would be great to be able to independently adjust above bridge speed so the whole print speed does not need to be slowed to overcome this issue.

SimonSolar2C, 2023-05-25:
Yes this can be done tediously but would be so nice as a tick box.  even a double bridge layer would work

SimonSolar2C, 2023-05-25:
I really need this now because I do nice bridge layers then the next layer catches it and clumps up and make a hole in my nice bridging.