supermerill/SuperSlicer #2903: Aligned rectilinear infill

cristianku, 2022-06-29:
Why there is not the aligned rectilinear in Superslicer as is present in Prusa slicer ?

neophyl, 2022-06-30:
Maybe because in Susi you can also set the increment angle of your infill.  Infill>Advanced>Angle>Increment.  With this you can stop it cross hatching like usual.  It also works on any infill pattern.  Experiment a little, see what you get.  Try setting it at 270.

stefa168, 2022-10-29:
Thanks, setting the increment to 270 did the trick for me!

VanessaE, 2023-08-19:
I know this is a little late, but I've been tinkering with that whole crystal clear prints thing, and the lack of a true aligned rectilinear infill mode has been standing in the way of that experiment.

Just changing the infill angle increment to 90 or 270 is NOT sufficient.  Sure, the infill lines will all be drawn at the same angle from one layer to the next, but they won't form a perfect stack in the Z direction -- the lines still wander sideways a bit from layer to layer, as the shape of the part changes.

However, just today I found a workaround, at least for version 2.5.59.0.  I'm sharing my findings here, as this issue seems to be the most appropriate:

* Set 0 top and 0 bottom solid infill layers
* Use the Honeycomb infill pattern (of all things!)
* Set your sparse infill density to 99.999%.  We do this because if you set it to 100%, the program will complain about it being inappropriate for Honeycomb infill.
* Set it to not connect the infill to the perimeters
* Use an initial rotation of 45° (or whatever floats your Benchy)
* Set a per-layer angle increment of either 120°.

I've put a screenshot below to show the full page of settings.

This combo of settings exploits a detail in how the Honeycomb pattern is drawn, causing its lines to come out perfectly straight (rather than hexagonal), and perfectly aligned layer to layer.  The result appears to be roughly equivalent to the old aligned rectilinear infill mode, at least in the models I've tried it with.

You could even turn on "connected" if you want, but while the the overall line pattern would be a little more equivalent to the old aligned mode, the connections/turnarounds jitter around a bit from layer to layer (see also #3734).  The bulk of the infill does remain aligned, though.

There are some caveats here:

1) Since this is technically the Honeycomb pattern, the infill lines are still divided into many small segments.  However, since they're aligned and perfectly straight anyway, they're drawn more or less identical to simple one-segment infill lines, so I don't think that'll cause any problem.  My printer seems to move perfectly smoothly as it draws them (very recent Marlin firmware)

2) The infill draw direction flips back and forth from one layer to the next.  That is to say, if a given infill line is drawn from northwest to southeast, that same infill line will probably be drawn from southeast to northwest in the next layer.  I don't know if this will have any impact on clear prints, since the infill lines still form a perfect stack.

3) This isn't monotonic infill -- individual sections of infill won't be drawn with their neighboring lines coordinated like monotonic infill normally goes, so light can still glint off of the top surfaces the way it used to with rectilinear infill.

[image]