supermerill/SuperSlicer #2907: SuperSlicer handling bridges as overhangs

MakManIfest, 2022-07-02:
### What happened?

I am printing some parts for a Voron 2.4 build and have come across an issue with superslicer.

Previous to using SS I have been using S3d so I am using my experience with S3d as a basis for what I perceive to be the issue.

I have run all the calibration tests to tune  SuperSlicer for the filament I am using, eSun PLA+ Black. I am using ver 2.3.57.12 of SS.

The part I am printing is a hinge, "bottom_panel_hinge_x2.stl"

The issue I am having is that SS perceives the first layer lines of the hinge pin as an overhang when it really is a bridge. This results in the part not printing correctly.

I have sliced, and printed to confirm, the part in S3d for comparison. The printed part in SS does not hinge. It binds as the hinge pin layers are not produced as required and weld to the layers below. The S3d part works as it should.

[image]
This is the layer the hinge pin starts. Without gap fill enabled. As you can see the hinge pin layer is identified as an overhang.

[image]
Same as above with gap fill enabled.

[image]
Here is the part sliced in S3d and correctly identified bridge section.

### Project file & How to reproduce

[bottom_panel_hinge_x2.zip](https://github.com/supermerill/SuperSlicer/files/9032801/bottom_panel_hinge_x2.zip)

### Version

2.3.57.12

### Operating system

Win 10

### Printer model

Voxelabs Aquila N32, HeroMe6, Dragonfly BMS, Orbiter1.5, Dual Z, Klipper

Henyckma-Sensei, 2022-07-04:
#2829 In my post I explained exactly the same thing, you can take a look at it

supermerill, 2022-07-12:
overhangs are unsupported periemter lines. They sometimes don't go in strait lines.
bridges are infill lines, and they always go in strait lines.

In older versions, both uses the bridge flow, speed, and accelecration. Now you can set them apart if you want, but it's not mandatory.

If you set the overhangs witht he same settigns as the bridges, then it's the same things.

If you really want to create infill and no overhangs, then you can try some "no perimeters on bridge areas" options.