supermerill/SuperSlicer #2909: [FR] Internal bridge infill: Activate/deactivate or configure like any other infill type

ldiegos, 2022-07-03:
**Is your feature request related to a problem? Please describe.**
Is not a problem, but actually it creates a problem on my printed models

**Describe the solution you'd like**
I need to know if it is feasible to have a checkbox to configure superslicer to avoid generating the "Internal Bridge infill" layer, because its breaks the final result with all transparent/translucent filaments the final result has a horrible view in the top layer, but beautiful in the bottom layers, because the "internal bridge infill" is not create in the bottom part.

**Describe how it would work**
I have some stl to be printed on transparent/translucent filaments, flex or pla and is so awesome to choose all the bottom layers as Archimedean chords and also to choose the infill sparse as Archimedean Chords. When the print is finished the bottom looks great:

Bottom Layers:
[image]

Infill: 
[image]

But, when the prints reach's the top layers, the final result is very poor, not visually acceptable, because the "internal bridge infill appears and breaks the infill visualization and mess it up:
[image]

After that the top layers appeared as Archimedean Chords, but seems to be disconnected from the overall result:
[image]

See this printed model:
Bottom view:
[image]

Top view:
[image]

**Describe alternatives you've considered**
I would like to have, a checkbox, in the infill section with the "internal bridge infil." on/off.
or
I would like to have also the same parameters to be configured like the rest of the layers type, a new row with "internal bridge infill" "Rectilinear/Concentric/Archimedean Chords/..." "Connected"

Both ideas sounds great for me.

[image]

**Additional context**
Add any other context or paint drawing about the feature request here.

Thanks a lot SuperMerill!
Luis

ubershy, 2023-12-27:
I also would like to have this feature. For now I have to use Cura for printing beautiful things with translucent filament.

supermerill, 2023-12-27:
I guess it's sensible for visual stuff.
I don't know if a circle can be printed reliably over sparse infill, i very doubt it.