supermerill/SuperSlicer #2911: Complete individual object sort order in Zig Zag pattern

LazaroFilm, 2022-07-05:
When Printing with multiple identical objets with `Complete individual objects:` checked on, and `Object sort:` set to `lowest Y`, it prints in a weird pattern in the X axis, which forces the carriage to travel back and forth more than needed. 
Ideally, the object order should be a Zig Zag pattern starting from the home position (0, 0) to avoid having the carriage travel over already printed parts uselessly and potentially knocking them over.

[image]

[image]

neophyl, 2022-07-06:
The normal option is to print in the tree order (right panel), with all your objects having the same y position on the same row then it may be sorting them within each row still based on the right panel order.  You can check this by using the e shortcut to display labels to see if it matches. 

If you had included the requested zipped up project file (3mf) then we would be able to check.  Without the required information then help will be limited and any suggestions will be based of pure guesswork.

LazaroFilm, 2022-07-10:
You may be right about the tree order for same Y height elements. 
[image]

Attached is the 3mf project file.
[Rod Clip 15 project file.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9079144/Rod.Clip.15.project.file.3mf.zip)

neophyl, 2022-07-11:
Yeah it looks like it doesnt it.  The order within each row precisely matches the object order.
If people really want this automated then I guess it also needs a sort by Y and then By X option adding.

supermerill, 2022-07-12:
a kind of "nearest" ?

LazaroFilm, 2022-08-13:
> A quick way would be to select column 2 and nudge all 4 pack a few mm, then row 3 a few more and finally row 4 slightly further. That way they would be placed in a zig zag and the x positions of each object on the row would no longer be identical.

Thanks! I found a way, by rotating all the objects at once by 0.001 degree on the Z axis it now makes them print all in one direction from left to right and from front to back. Not quite the zig zag pattern I would love but better, but now all my parts are rotated by 0.01 which isn't great either...