supermerill/SuperSlicer #2921: Filament temperature calibration slice deformation object

Shahriar57, 2022-07-11:
### Version

Version: 2.3.57.12
Build: SuperSlicer_2.3.57.1

### Operating system type + version
Windows Version: Windows 10 (build 19044), 64-bit edition

### Behavior
* when slice the calibration temp object deformation check pic
* it is not diffrence if i change nozzle diameter to even 0.4 mm result same
* i load .3mf file in lastest pursa slicer and it is slice normal but in superslicer the result same as pic

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here
[Temperature calibration.zip](https://github.com/supermerill/SuperSlicer/files/9082755/Temperature.calibration.zip)
[image]
_

neophyl, 2022-07-11:
Its something in your curve smoothing settings (which is something Prusa Slicer doesn't have) that is causing the issue.  Setting them to the defaults makes it slice correctly.

Shahriar57, 2022-07-11:
Thank you, curve smoothing disable and working fine

closed 2022-07-11 by Shahriar57