supermerill/SuperSlicer #2922: Z-lift does not make any difference in G-code or can't be deactivated

Z-oid, 2022-07-11:
### Version
2.3.57

### Operating system type + version
Windows 10 Pro 10.0.19043 Build 19043

### Behavior
Printing on a MK3s Fysetc clone
For test rints I wanted to deactivate the Z-lift in the slicer. When I enter the value of 0 mm for the Z-lift I still have Z-lifts in the generated G-code.
If I change the value to 10 mm, to make a Z-lift very clear, generate the code and print, I get the same results like in the print before. So making a change in the slicer is not affecting the G-code at all.

#### Project File (.3MF) where problem occurs
I attached the project and gcode files 
[Z-Lift 0 mm and 10 mm.zip](https://github.com/supermerill/SuperSlicer/files/9083035/Z-Lift.0.mm.and.10.mm.zip)

neophyl, 2022-07-11:
Your filament profile has a 0.2mm z lift enabled in the over ride section.  As such it is (correctly) replacing whatever you have set under the extruder.  Same as the retraction in the filament is set to 1mm which is replacing that set under the extruder.

As such no bug here.

Z-oid, 2022-07-11:
I was not aware of this section. In that case the problem is solved. Thanks a lot for your help.

closed 2022-07-12 by supermerill