supermerill/SuperSlicer #2924: Perimeter of layer starts with a bridge and too little "anchor"

alexvonsiebenthal, 2022-07-11:
**Bug description**
I have a model that is basically a cylindrical grid. It has a lot of bridges that work well. However, the start of the perimeter on the layer is right on the bridge rather than on a solid layer below, which does absolutely not make sense to me. It often fails to attach in the print and it sags down.
[image]

[image]

**Expected behavior**
The slicer should either determine a better starting point (like in the middle of one of the pillars) or let me define a minimum "anchor distance" for a bridge.

[BridgeStartBug.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9087552/BridgeStartBug.3mf.zip)

**To Reproduce**
See project file attached

**Screenshots**
See screenshots above

**Desktop (please complete the following information):**
macOS, SS 2.4.58.3, Voron 2.4 / Stealthburner

neophyl, 2022-07-12:
If you change your seam position to Corners it does slice with those initial perimeters on top of the layer below.  It also puts more on the loop back when it changes to the next perimeter.  Same with Nearest.  
I think your choice of aligned is what is causing the problem.

alexvonsiebenthal, 2022-07-12:
Thank you! The SS version in use seems to have a bug where I cannot select the seam position at all. And apparently "Nearest" is already selected.

[image]

neophyl, 2022-07-12:
Yeah I have 2.5.58.2 on my machine currently.