supermerill/SuperSlicer #2925: Superslicer Post processing scripts looking at wrong directory when launched from file

DatenThielt, 2022-07-12:
### What happened?

When launching Superslicer from an STL file that is not in the programs root directory, prost processing scripts are looking at the directory of the file instead of the directory of superslicer.exe.  When trying to generate the GCode this will cause an error and the gcode generation will fail.

### Project file & How to reproduce

Make sure you have a post processing script or application such as preprocess_cancellation.exe; in your print settings

Launch SuperSlicer from an STL for example from C:/User/Desktop
Try to slice and generate gcode.  Superslicer will error and say it cannot find the postprocessing script as it is looking for preprocess_cancellation.exe on the C:/User/Desktop location.

Potential workaround would be to hardcode the path in the processing script

### Version

2.4.58.2

### Operating system

Windoows 11

### Printer model

Voron 2.4

DatenThielt, 2022-07-12:
So currently all of my project files and STLS sit in C:\User\3d Object\My Projects\Project name (And various sub directories for each part type etc),  

If I double click my STL's  set them up for printing and then try to generate gcode to be sent, It gives me the error.  Since I have about 16 different projects I don't want 16 copies of the post processing script to reside in the directory of the STL's.  I also cant see why anyone would want the post processing script to be run from the same directory as their STL's.

I definitely do not think enforcing post processing to be in the same directory as superslicer.exe is a good idea and full length directories should always take priority (aka specifying C:/Program Files/SuperSlicer/preprocess_cancellation.exe), however just launching SS from an STL should still load the postprocessing from the root directory and not be trying to load it from the STL Directory  or at a minimum explain where it is trying to launch the post processing file from so that its clear to the user what the application is trying todo.  Currently it is very unclear as to why it could not generate the gcode (see attached)

Hopefully I am making sense :)

[image]

supermerill, 2022-07-18:
script file will be searched from different directory if not found on the current on
- from the exe directory
- from the configuration directory
- from the user directory

DatenThielt, 2022-07-18:
Amazing!

closed 2024-02-11 by supermerill