supermerill/SuperSlicer #2928: [macOS] (nightly) crash when adding object modifiers

alexvonsiebenthal, 2022-07-12:
### What happened?

When I want to have object-specific modifies (i.e. flow, line width etc.), the application crashes.
[image]

### Project file & How to reproduce

Crash report attached
[SuSi Crash Report.txt](https://github.com/supermerill/SuperSlicer/files/9095735/SuSi.Crash.Report.txt)

### Version

2.4.58.3 nigthly

### Operating system

macOS 12.3.1

### Printer model

voron 2.4 / stealthburner

alexvonsiebenthal, 2022-07-14:
it happens on any project. Maybe because it's ARM?

alexvonsiebenthal, 2022-07-14:
Project + Video:
[Archive.zip](https://github.com/supermerill/SuperSlicer/files/9110480/Archive.zip)

supermerill, 2022-07-14:
It's an arm. Still can't replicate
[Screen Recording 2022-07-14 at 22.35.49.zip](https://github.com/supermerill/SuperSlicer/files/9112252/Screen.Recording.2022-07-14.at.22.35.49.zip)

supermerill, 2022-07-15:
@alexvonsiebenthal see #2055