supermerill/SuperSlicer #2933: Decelerate with target acceleration doesn't change gcode

cfytrok, 2022-07-14:
### What happened?

Decelerate with target acceleration doesn't change acceleration. 
Expected:
```
G1 Z0.32 F30000 ; move to next layer (0)
; custom gcode: layer_gcode
;AFTER_LAYER_CHANGE
;0.32
; custom gcode end: layer_gcode

G92 E0 ; reset extruder position to flush any extruder axis rounding
M204 P1200 T1200 ; adjust acceleration
G1 X20.21 Y90.21 ; move to first External perimeter point (minimum acceleration)
;TYPE:External perimeter
;WIDTH:0.42
G1 F1200
G1 X39.79 Y90.21 E0.91518 ; External perimeter
G1 X39.79 Y109.79 E0.91518 ; External perimeter
G1 X20.21 Y109.79 E0.91518 ; External perimeter
G1 X20.21 Y90.21 E0.91518 ; External perimeter
;WIPE_START
G1 X20.556 Y90.41 F30000 ; move inwards before travel
;WIPE_END
; acceleration to travel
M204 T5000
G1 X100.922 Y90.249 ; move to first External perimeter point (acceleration)
; decel to extrusion
M204 T1200
G1 X120.21 Y90.21 ; move to first External perimeter point (deceleration)
; end travel
G1 F1200
G1 X139.79 Y90.21 E0.91518 ; External perimeter
G1 X139.79 Y109.79 E0.91518 ; External perimeter
G1 X120.21 Y109.79 E0.91518 ; External perimeter
G1 X120.21 Y90.21 E0.91518 ; External perimeter
;WIPE_START
G1 X120.556 Y90.41 F30000 ; move inwards before travel
;WIPE_END
M106 S255 ; enable fan
;LAYER_CHANGE
```
Got:
```
G1 Z0.32 F30000 ; move to next layer (0)
; custom gcode: layer_gcode
;AFTER_LAYER_CHANGE
;0.32
; custom gcode end: layer_gcode

G92 E0 ; reset extruder position to flush any extruder axis rounding
M204 P1200 T1200 ; adjust acceleration
G1 X20.21 Y90.21 ; move to first External perimeter point (minimum acceleration)
;TYPE:External perimeter
;WIDTH:0.42
G1 F1200
G1 X39.79 Y90.21 E0.91518 ; External perimeter
G1 X39.79 Y109.79 E0.91518 ; External perimeter
G1 X20.21 Y109.79 E0.91518 ; External perimeter
G1 X20.21 Y90.21 E0.91518 ; External perimeter
;WIPE_START
G1 X20.556 Y90.41 F30000 ; move inwards before travel
;WIPE_END
; acceleration to travel
G1 X100.922 Y90.249 ; move to first External perimeter point (acceleration)
; decel to extrusion
G1 X120.21 Y90.21 ; move to first External perimeter point (deceleration)
; end travel
G1 F1200
G1 X139.79 Y90.21 E0.91518 ; External perimeter
G1 X139.79 Y109.79 E0.91518 ; External perimeter
G1 X120.21 Y109.79 E0.91518 ; External perimeter
G1 X120.21 Y90.21 E0.91518 ; External perimeter
;WIPE_START
G1 X120.556 Y90.41 F30000 ; move inwards before travel
;WIPE_END
M106 S255 ; enable fan
;LAYER_CHANGE
```

### Project file & How to reproduce

[deceleration_test.zip](https://github.com/supermerill/SuperSlicer/files/9110186/deceleration_test.zip)

### Version

2.4.58.3

### Operating system

Windows 10

### Printer model

Ender 3

macchinette, 2023-02-10:
Salve, io ho un problema analogo sempre riferito alla gestione delle accellerazioni, vedi link.
https://github.com/supermerill/SuperSlicer/issues/1649

cfytrok, 2023-08-26:
The problem seems to be in this code. It does not check whether the m_current_travel_acceleration has changed.
https://github.com/supermerill/SuperSlicer/blob/45d0532845b63cd5cefe7de7dc4ef0e0ed7e030a/src/libslic3r/GCodeWriter.cpp#L268