supermerill/SuperSlicer #2936: Unable to Launch SuperSlicer AppImage

m-reiner, 2022-07-14:
### Version
SuperSlicer-ubuntu_18.04-2.4.58.3.AppImage

### Operating system type + version
Fedora 36 Workstation

### Behavior
When attempting to run SuperSlicer-ubuntu_18.04-2.4.58.3.AppImage, it never launches and I see the following errors:

``(process:109451): Gtk-CRITICAL **: 15:29:40.984: _gtk_css_lookup_resolve: assertion '(((__extension__ ({ GTypeInstance *__inst = (GTypeInstance*) ((provider)); GType __t = ((_gtk_style_provider_private_get_type ())); gboolean __r; if (!__inst) __r = (0); else if (__inst->g_class && __inst->g_class->g_type == __t) __r = (!(0)); else __r = g_type_check_instance_is_a (__inst, __t); __r; }))))' failed``

``(process:109451): GLib-GObject-CRITICAL **: 15:29:40.984: g_object_set_data_full: assertion 'G_IS_OBJECT (object)' failed``

``(process:109451): Gtk-ERROR **: 15:29:40.984: Can't create a GtkStyleContext without a display connection``

m-reiner, 2022-07-15:
I managed to compile SuperSlicer from source after figuring out what the Fedora equivalent dependency packages are, but when I tried to launch the version I compiled locally I wound up with the same errors.  Same goes for every version listed on your GitHub (SuperSlicer-ubuntu_18.04-2.4.58.3.AppImage, SuperSlicer-ubuntu_18.04-gtk2-2.4.58.3.AppImage, SuperSlicer_2.4.58.3_linux64_220714.tar.zip, and the version I compiled).

I should note that I was able to download the latest version of PrusaSlicer (PrusaSlicer-2.4.2+linux-x64-GTK3-202204251120.AppImage) from their GitHub page, which launches with zero issues.

m-reiner, 2022-07-15:
Another update: the version I had been using up until this point (I renamed the file so I honestly don't know the version anymore, but it's a few versions back) is also causing these same errors now and will not launch.  I suspect something updated in Fedora that caused something else to break for SuperSlicer.  Again, though, I can launch PrusaSlicer's most recent release with no issues.

sdooweloc, 2022-07-15:
I am having an identical experience on Arch. Some dependency probably
changed and broke everything but since logging gives nothing I can't even
begin to imagine where the break is.

supermerill, 2022-07-15:
Don't have any problem launching it on ubuntu 22.04
Sadly, i can't test every linux distribution.
If you find the fix, i can merge it.

m-reiner, 2022-07-15:
Totally respect that you can't test all the distros, though I've seen similar issues raised by Fedora and CentOS users, so this might extend to anyone using RPM packages.

I'm probably not technical enough to figure it out, but I'll do some digging and if I figure it out I will definitely let you know.

On a side note: thanks for all the hard work you put into SuperSlicer; it's my favorite for many good reasons!

m-reiner, 2022-07-15:
Oh, and if you or anyone else decides to also test this out on an RPM distro, here are the changes I made to the BuildLinux.sh to be able to compile:

Lines 5 and 6 as well as 86 and 87:
``FOUND_GTK2_DEV=$(yum list gtk* | grep gtk2-dev)``
``FOUND_GTK3_DEV=$(yum list gtk* | grep gtk3-dev)``

Lines 71 and 72:
``        echo -e "\nInstalling: gtk2-devel glew-devel systemd-devel dbus-devel cmake git\n"``
``        dnf install gtk2-devel libglew-dev systemd-devel dbus-devel cmake git``

Lines 74 and 75:
``        echo -e "\nFind gtk3-devel, installing: gtk3-devel glew-devel systemd-devel dbus-devel cmake git\n"``
``        dnf install gtk3-devel libglew-dev systemd-devel dbus-devel cmake git``

olby, 2022-07-20:
The last build that works for me is: 65f41c8
The build after that (1b97666) fail with the error in the issue description.

supermerill, 2022-07-20:
@olby
can you test the two test builds From the action menu?

m-reiner, 2022-07-20:
I tested out 'test build 1' but it failed with the following errors when I went to run it after building:

``(process:1119722): Gtk-CRITICAL **: 19:55:46.489: _gtk_css_lookup_resolve: assertion '(((__extension__ ({ GTypeInstance *__inst = (GTypeInstance*) ((provider)); GType __t = ((_gtk_style_provider_private_get_type ())); gboolean __r; if (!__inst) __r = (0); else if (__inst->g_class && __inst->g_class->g_type == __t) __r = (!(0)); else __r = g_type_check_instance_is_a (__inst, __t); __r; }))))' failed``

``(process:1119722): GLib-GObject-CRITICAL **: 19:55:46.489: g_object_set_data_full: assertion 'G_IS_OBJECT (object)' failed``

``(process:1119722): Gtk-ERROR **: 19:55:46.489: Can't create a GtkStyleContext without a display connection``

I tested out 'test build 2' but it failed with the following errors when I went to run it after building:

``(process:1119225): Gtk-CRITICAL **: 19:53:37.254: _gtk_css_lookup_resolve: assertion '(((__extension__ ({ GTypeInstance *__inst = (GTypeInstance*) ((provider)); GType __t = ((_gtk_style_provider_private_get_type ())); gboolean __r; if (!__inst) __r = (0); else if (__inst->g_class && __inst->g_class->g_type == __t) __r = (!(0)); else __r = g_type_check_instance_is_a (__inst, __t); __r; }))))' failed``

``(process:1119225): GLib-GObject-CRITICAL **: 19:53:37.254: g_object_set_data_full: assertion 'G_IS_OBJECT (object)' failed``

``(process:1119225): Gtk-ERROR **: 19:53:37.254: Can't create a GtkStyleContext without a display connection
Trace/breakpoint trap (core dumped)``

m-reiner, 2022-07-21:
I also tried testing https://github.com/supermerill/SuperSlicer/commit/65f41c83000dc3e9c54656fe074f1963fd4e34ba per @olby's earlier post, but that gave me the same errors:

``(process:1305191): Gtk-CRITICAL **: 20:34:02.399: _gtk_css_lookup_resolve: assertion '(((__extension__ ({ GTypeInstance *__inst = (GTypeInstance*) ((provider)); GType __t = ((_gtk_style_provider_private_get_type ())); gboolean __r; if (!__inst) __r = (0); else if (__inst->g_class && __inst->g_class->g_type == __t) __r = (!(0)); else __r = g_type_check_instance_is_a (__inst, __t); __r; }))))' failed`

``(process:1305191): GLib-GObject-CRITICAL **: 20:34:02.399: g_object_set_data_full: assertion 'G_IS_OBJECT (object)' failed``

``(process:1305191): Gtk-ERROR **: 20:34:02.399: Can't create a GtkStyleContext without a display connection``

olby, 2022-07-21:
Sadly both test build 1 and 2 fail in the same way :(

veehexx, 2022-07-23:
looks like still the same issue for me for TestBuild 3-6 (Fedora36, updated from repos about 10 minutes ago). I use --datadir for my 'production' SuSi config, so these should be first-time run experiences: 

> [USER@fedora Downloads]$ ./SuperSlicer-gtk3.AppImage-**tb3**/SuperSlicer_ubu64.AppImage
> [2022-07-23 20:43:42.137030] [0x00007f6907bd6540] [trace]   Initializing StaticPrintConfigs
> SuperSlicer GUI initialization failed: boost::filesystem::copy_file: File exists: "/tmp/.mount_SuperSxbWLVU/bin/../resources/ui_layout/example/colors.ini", "/home/USER/.config/SuperSlicer/ui_layout/example/colors.ini"
> 
> (process:21249): Gtk-CRITICAL **: 20:43:42.224: _gtk_css_lookup_resolve: assertion '(((__extension__ ({ GTypeInstance *__inst = (GTypeInstance*) ((provider)); GType __t = ((_gtk_style_provider_private_get_type ())); gboolean __r; if (!__inst) __r = (0); else if (__inst->g_class && __inst->g_class->g_type == __t) __r = (!(0)); else __r = g_type_check_instance_is_a (__inst, __t); __r; }))))' failed
> 
> (process:21249): GLib-GObject-CRITICAL **: 20:43:42.224: g_object_set_data_full: assertion 'G_IS_OBJECT (object)' failed
> 
> (process:21249): Gtk-ERROR **: 20:43:42.224: Can't create a GtkStyleContext without a display connection
> Trace/breakpoint trap (core dumped)
---
> [USER@fedora Downloads]$ ./SuperSlicer-gtk3.AppImage-**tb4**/SuperSlicer_ubu64.AppImage
> [2022-07-23 20:43:47.408540] [0x00007f3edc04e540] [trace]   Initializing StaticPrintConfigs
> SuperSlicer GUI initialization failed: boost::filesystem::copy_file: File exists: "/tmp/.mount_SuperS7Vil7M/bin/../resources/ui_layout/example/colors.ini", "/home/USER/.config/SuperSlicer/ui_layout/example/colors.ini"
> 
> (process:21274): Gtk-CRITICAL **: 20:43:47.498: _gtk_css_lookup_resolve: assertion '(((__extension__ ({ GTypeInstance *__inst = (GTypeInstance*) ((provider)); GType __t = ((_gtk_style_provider_private_get_type ())); gboolean __r; if (!__inst) __r = (0); else if (__inst->g_class && __inst->g_class->g_type == __t) __r = (!(0)); else __r = g_type_check_instance_is_a (__inst, __t); __r; }))))' failed
> 
> (process:21274): GLib-GObject-CRITICAL **: 20:43:47.498: g_object_set_data_full: assertion 'G_IS_OBJECT (object)' failed
> 
> (process:21274): Gtk-ERROR **: 20:43:47.498: Can't create a GtkStyleContext without a display connection
> Trace/breakpoint trap (core dumped)
---

> [USER@fedora Downloads]$ ./SuperSlicer-gtk3.AppImage-**tb5**/SuperSlicer_ubu64.AppImage
> [2022-07-23 20:43:51.417613] [0x00007f0c532a1540] [trace]   Initializing StaticPrintConfigs
> SuperSlicer GUI initialization failed: boost::filesystem::copy_file: File exists: "/tmp/.mount_SuperS8x6NHX/bin/../resources/ui_layout/example/colors.ini", "/home/USER/.config/SuperSlicer/ui_layout/example/colors.ini"
> 
> (process:21297): Gtk-CRITICAL **: 20:43:51.500: _gtk_css_lookup_resolve: assertion '(((__extension__ ({ GTypeInstance *__inst = (GTypeInstance*) ((provider)); GType __t = ((_gtk_style_provider_private_get_type ())); gboolean __r; if (!__inst) __r = (0); else if (__inst->g_class && __inst->g_class->g_type == __t) __r = (!(0)); else __r = g_type_check_instance_is_a (__inst, __t); __r; }))))' failed
> 
> (process:21297): GLib-GObject-CRITICAL **: 20:43:51.500: g_object_set_data_full: assertion 'G_IS_OBJECT (object)' failed
> 
> (process:21297): Gtk-ERROR **: 20:43:51.500: Can't create a GtkStyleContext without a display connection
> Trace/breakpoint trap (core dumped)

---
> [USER@fedora Downloads]$ ./SuperSlicer-gtk3.AppImage-**tb6**/SuperSlicer_ubu64.AppImage
> [2022-07-23 20:43:55.012023] [0x00007fafe22a9540] [trace]   Initializing StaticPrintConfigs
> SuperSlicer GUI initialization failed: boost::filesystem::copy_file: File exists: "/tmp/.mount_SuperSTCo7xV/bin/../resources/ui_layout/example/colors.ini", "/home/USER/.config/SuperSlicer/ui_layout/example/colors.ini"
> 
> (process:21320): Gtk-CRITICAL **: 20:43:55.106: _gtk_css_lookup_resolve: assertion '(((__extension__ ({ GTypeInstance *__inst = (GTypeInstance*) ((provider)); GType __t = ((_gtk_style_provider_private_get_type ())); gboolean __r; if (!__inst) __r = (0); else if (__inst->g_class && __inst->g_class->g_type == __t) __r = (!(0)); else __r = g_type_check_instance_is_a (__inst, __t); __r; }))))' failed
> 
> (process:21320): GLib-GObject-CRITICAL **: 20:43:55.106: g_object_set_data_full: assertion 'G_IS_OBJECT (object)' failed
> 
> (process:21320): Gtk-ERROR **: 20:43:55.106: Can't create a GtkStyleContext without a display connection
> Trace/breakpoint trap (core dumped)

supermerill, 2022-07-23:
@olby
test build 3-6 are between the 65f41c83000dc3e9c54656fe074f1963fd4e34ba and the test build 1, can you say the first one that doesn't work anymore?

veehexx, 2022-07-24:
this might be fixed with supermerill's [commit](https://github.com/supermerill/SuperSlicer/commit/25751573af5dbd8b6f973a3d13b873bfdeda3788) a few hours ago. 
build that works for me: https://github.com/supermerill/SuperSlicer/actions/runs/2727878113

other bug ref that appears to be related to this: https://github.com/supermerill/SuperSlicer/issues/2938

m-reiner, 2022-07-25:
Thanks for sharing, can confirm that the AppImage of the build you linked (SuperSlicer_2.5.59.0_2022-07-24, https://github.com/supermerill/SuperSlicer/actions/runs/2727878113) does work for me!

I'm not convinced that my issue and the issue referenced in https://github.com/supermerill/SuperSlicer/issues/2938 are the same, though.  I don't see a Segmentation Fault error when I try to run the builds with which I was having issues, and PrusaSlicer worked just fine where the tag on https://github.com/supermerill/SuperSlicer/issues/2938 lists the same problem existing in PrusaSlicer.

Either way, thanks so much for the heads up!  No more Windows version under Wine crashing every five minutes!

olby, 2022-07-29:
> [65f41c8](https://github.com/supermerill/SuperSlicer/commit/65f41c83000dc3e9c54656fe074f1963fd4e34ba)

Nightly ubuntu-GTK3 #126 65f41c8 nightly_dev  (Works)
test build 3 C/C++ Nightly ubuntu-GTK3 #132: Commit 495b206 (Broken)
test build 4 C/C++ Nightly ubuntu-GTK3 #133: Commit 9147a2a (Broken)
test build 5 C/C++ Nightly ubuntu-GTK3 #135: Commit 6b26237 (Broken)
test build 6 C/C++ Nightly ubuntu-GTK3 #136: Commit fabcaf4 (Broken)

olby, 2022-07-29:
test build 3 C/C++ Nightly ubuntu-GTK3 #132: Commit 495b206 (Broken)
```
#0  g_log_writer_default (log_level=<optimized out>, fields=0x7fffffff35d0, n_fields=6, user_data=0x0) at ../glib/gmessages.c:557
#1  0x00007ffff6d16861 in g_log_structured_array (n_fields=6, fields=0x7fffffff35d0, log_level=G_LOG_LEVEL_ERROR) at ../glib/gmessages.c:1973
#2  g_log_structured_array (log_level=G_LOG_LEVEL_ERROR, fields=0x7fffffff35d0, n_fields=6) at ../glib/gmessages.c:1946
#3  0x00007ffff6d16ab2 in g_log_structured_standard
    (log_domain=log_domain@entry=0x7ffff7652060 "Gtk", log_level=log_level@entry=G_LOG_LEVEL_ERROR, file=file@entry=0x7ffff76a9098 "../gtk/gtkstylecontext.c", line=line@entry=0x7ffff766bdff "348", func=func@entry=0x7ffff76aa760 <__func__.70> "gtk_style_context_init", message_format=message_format@entry=0x7ffff76a9a08 "Can't create a GtkStyleContext without a display connection") at ../glib/gmessages.c:2030
#4  0x00007ffff74c93c9 in gtk_style_context_init (context=0x555558334c70) at ../gtk/gtkstylecontext.c:348
#5  0x00007ffff7a94fd9 in g_type_create_instance (type=0x555558338070 [None]) at ../gobject/gtype.c:1929
#6  0x00007ffff7a7ce38 in g_object_new_internal (class=0x5555583381e0, params=params@entry=0x0, n_params=n_params@entry=0) at ../gobject/gobject.c:2011
#7  0x00007ffff7a7e081 in g_object_new_with_properties (object_type=0x555558338070 [None], n_properties=0, names=names@entry=0x0, values=values@entry=0x0)
    at ../gobject/gobject.c:2181
#8  0x00007ffff7a7eb21 in g_object_new (object_type=<optimized out>, first_property_name=<optimized out>) at ../gobject/gobject.c:1821
#9  0x00007ffff75df9b3 in _gtk_style_new_for_path (screen=0x0, path=path@entry=0x5555580fc220) at ../gtk/deprecated/gtkstyle.c:854
#10 0x00007ffff75dfa32 in gtk_style_new () at ../gtk/deprecated/gtkstyle.c:888
#11 0x00007ffff75dfac1 in gtk_widget_get_default_style () at ../gtk/deprecated/gtkstyle.c:4061
#12 gtk_widget_get_default_style () at ../gtk/deprecated/gtkstyle.c:4050
#13 0x00007ffff756afb5 in gtk_widget_init (instance=<optimized out>, g_class=0x555558331da0) at ../gtk/gtkwidget.c:4479
#14 0x00007ffff7a94f9b in g_type_create_instancePython Exception <class 'TypeError'>: can only concatenate str (not "NoneType") to str
 (type=) at ../gobject/gtype.c:1923
#15 0x00007ffff7a7ce38 in g_object_new_internal (class=class@entry=0x555558331da0, params=params@entry=0x7fffffff40e0, n_params=n_params@entry=3)
    at ../gobject/gobject.c:2011
#16 0x00007ffff7a7e5e5 in g_object_new_valist
    (object_type=<optimized out>, first_property_name=first_property_name@entry=0x7ffff7655391 "use-header-bar", var_args=var_args@entry=0x7fffffff43b0)
    at ../gobject/gobject.c:2355
#17 0x00007ffff7a7eafd in g_object_newPython Exception <class 'TypeError'>: can only concatenate str (not "NoneType") to str
 (object_type=object_type@entry=, first_property_name=first_property_name@entry=0x7ffff7655391 "use-header-bar")
    at ../gobject/gobject.c:1824
#18 0x00007ffff742a080 in gtk_message_dialog_new
    (parent=<optimized out>, flags=GTK_DIALOG_MODAL, type=GTK_MESSAGE_ERROR, buttons=GTK_BUTTONS_OK, message_format=0x5555577cc14f "%s")
    at ../gtk/gtkmessagedialog.c:615
#19 0x00005555570436a7 in wxMessageDialog::GTKCreateMsgDialog() ()
#20 0x0000555557044bd0 in wxMessageDialog::ShowModal() ()
#21 0x0000555556fc8868 in wxMessageBox(wxString const&, wxString const&, long, wxWindow*, int, int) ()
#22 0x0000555555ef944e in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#23 0x00005555557a6638 in Slic3r::CLI::run(int, char**) ()
#24 0x000055555575b8fa in main ()
```

olby, 2022-07-29:
Hm, started 495b206 with a break point in gdb and copied some resources;

```
cp -r /tmp/.mount_SuperSC0okjm/resources/ui_layout/* ~/.config/SuperSlicer/ui_layout/
```

and then continued execution and now it starts and after this I can run it the normal way.

The bug is back as soon as i remove `~/.config/SuperSlicer` completely.

olby, 2022-07-29:
The proper fix is in
```
 test for #2938 C/C++ Nightly ubuntu-GTK3 #139: Commit 2575157
```
After restoring the the failing config and executing `test for #2938` the previously failing builds work.

m-reiner, 2022-08-11:
I can confirm that both the most recent stable release (Not arachne edition 2.4.58.4) and the newest pre-release (2.5.59.0 Arachne edition) both launch without this issue now.

Thank you to everyone for your help; you rock!

closed 2022-08-11 by m-reiner