supermerill/SuperSlicer #2937: Send Gcode To Printer Button Blinking

Imburr, 2022-07-14:
### What happened?

Not really a problem, but I assume it is not intended to flash like this.
When you mouseover the top left corner of the send to printer G button, the button blinks rapidly even if you do not continue to move the mouse.

### Project file & How to reproduce

https://user-images.githubusercontent.com/2759346/179073707-29b82c65-5f52-4ac5-9c1a-7b2b8f26c2bd.mp4

### Version

2.4.58.3

### Operating system

Windows 11

### Printer model

Ender 3 V2

DARKFiB3R, 2022-12-06:
Same here. Quite annoying tbh

Didn't realise it was only the top left corner of the button behaving like this. Easy to avoid now, knowing this.

Thanks.

neophyl, 2022-12-07:
I think that's the tooltip pop up appearing and disappearing rapidly.  I suspect when its not drawn and when it is drawn alters the activation area and when your mouse is in a particular spot it appears, which alters the activation area, which makes it go away, which makes it come back cyclically.