supermerill/SuperSlicer #2938: Superslicer 2.4.58.2-3 crashes upon launch (Segmentation fault)

d-avery, 2022-07-14:
### What happened?

I'm using Arch Linux with the Linux 5.18.11 kernel.

Used to have the Superslicer version from the AUR installed. I just switched to the official one to update and ever since I can't get it to start, all it shows is "Segmentation Fault (core dumped)" if I launch it from the terminal. 

I don't know what else I can tell you. On the AUR version all I did was set up my configuration. I renamed the ~/.config/SuperSlicer folder to check if that might have been the cause, but it segfaulted nonetheless.

### Project file & How to reproduce

Steps I took:
1: Get Superslicer 2.4.58.2-3 from the official Arch Linux community repository.
2: Launch it from the terminal using `superslicer`.
3: See segmentation fault error.

### Version

2.4.58.2-3

### Operating system

Arch Linux (Linux kernel 5.18.11)

### Printer model

Not relevant

sdooweloc, 2022-07-15:
I also have this exact same issue also with the appimage

`❯ superslicer -loglevel 5
[2022-07-14 20:13:21.380016] [0x00007fb436e68d80] [debug]   full path: 7714641723692594085
[2022-07-14 20:13:21.380029] [0x00007fb436e68d80] [debug]   single instance: undefined. other params: superslicer;-loglevel;5
[2022-07-14 20:13:21.380032] [0x00007fb436e68d80] [debug]   full lock path: /home/cole/.config/SuperSlicer/cache/7714641723692594085.lock
[2022-07-14 20:13:21.380042] [0x00007fb436e68d80] [debug]   Creating lockfile.
[2022-07-14 20:13:21.380045] [0x00007fb436e68d80] [info]    Instance check: Another instance not found or single-instance not set.
[2022-07-14 20:13:21.407709] [0x00007fb436e68d80] [trace]   System language detected (user locales and such): en_US
[1]    9643 segmentation fault (core dumped)  superslicer -loglevel 5
    ~                                                                                                            SEGV ✘
❯ Downloads/SuperSlicer-ubuntu_18.04-2.4.58.3.AppImage
[2022-07-14 20:16:14.269301] [0x00007fa4d954f480] [trace]   Initializing StaticPrintConfigs
SuperSlicer GUI initialization failed: boost::filesystem::copy_file: File exists: "/tmp/.mount_SuperSE7l1ca/bin/../resources/ui_layout/example/colors.ini", "/home/cole/.config/SuperSlicer/ui_layout/example/colors.ini"

(process:10480): Gtk-CRITICAL **: 20:16:14.345: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(process:10480): Gtk-CRITICAL **: 20:16:14.345: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(process:10480): Gtk-CRITICAL **: 20:16:14.345: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
[1]    10480 segmentation fault (core dumped)  Downloads/SuperSlicer-ubuntu_18.04-2.4.58.3.AppImage
    ~                                                                                                            SEGV ✘
❯ Downloads/SuperSlicer-ubuntu_18.04-2.4.58.3.AppImage -loglevel 5
[2022-07-14 20:16:25.009500] [0x00007feaa637a480] [trace]   Initializing StaticPrintConfigs
SuperSlicer GUI initialization failed: boost::filesystem::copy_file: File exists: "/tmp/.mount_SuperSicWWKn/bin/../resources/ui_layout/example/colors.ini", "/home/cole/.config/SuperSlicer/ui_layout/example/colors.ini"

(process:10502): Gtk-CRITICAL **: 20:16:25.085: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(process:10502): Gtk-CRITICAL **: 20:16:25.086: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(process:10502): Gtk-CRITICAL **: 20:16:25.086: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
[1]    10502 segmentation fault (core dumped)  Downloads/SuperSlicer-ubuntu_18.04-2.4.58.3.AppImage -loglevel 5`

arnej, 2022-07-15:
I have the same issue since last night on Arch Linux.

I rolled back the affected packages with the following command:

`sudo pacman -U /var/cache/pacman/pkg/wxgtk3-3.0.5.1-3-x86_64.pkg.tar.zst /var/cache/pacman/pkg/wxgtk-common-3.0.5.1-3-x86_64.pkg.tar.zst /var/cache/pacman/pkg/kicad-6.0.6-1-x86_64.pkg.tar.zst /var/cache/pacman/pkg/python-wxpython-1:4.0.7.2-3-x86_64.pkg.tar.zst /var/cache/pacman/pkg/superslicer-1:2.4.58.2-2-x86_64.pkg.tar.zst`

You can leave out kicad if you don't have it installed, of course.
After this, superslicer is at least usable again for me until this issue gets fixed properly.

I also think that this is already fixed in the latest version of superslicer and would also be fixed in Arch and it derivatives when they upgrade superslicer to the latest version (2.4.58.3).

arnej, 2022-07-15:
No, it's not because of superslicer (at least I think so).
It's because Arch updated wxwidgets and related packages yesterday, which also broke prusaslicer for me.

gyordanov, 2022-07-15:
SS does indeed segfault after the arch updates, but the latest prusaslicer 2.5 appimage works just fine ...

supermerill, 2022-07-15:
prusa updated many libraries for 2.5 (but not wxwidget), maybe it's why it still works.

sdooweloc, 2022-07-15:
> I have the same issue since last night on Arch Linux.
>
> I rolled back the affected packages with the following command:
>
> `sudo pacman -U /var/cache/pacman/pkg/wxgtk3-3.0.5.1-3-x86_64.pkg.tar.zst [...]

I tried rolling these back and had to delete the contents of both /usr/share/SuperSlicer/ui_layout/example/ and its counterpart in my .config directory. But this at least got me up and running for now.

veehexx, 2022-07-16:
same on Fedora36 using the appimage.
unfortunately rolling back to the previous version also has the same effect even after renaming the obvious stuff like .../SuperSlicer/ui_layout/ default&examples which are referenced in the segfault.

i don't appear to have anything 'wx' related installed (no wxwidgets, no wxgtk...)

down & out for me :(
obviously a proper fix is good but any suggestions on how to get the appimage (current or SuperSlicer-ubuntu_18.04-2.4.58.2) back working would be good in the meantime.

cowboy3d, 2022-07-18:
Same for me  on EndeavourOS - if I delete directories resources/ui_layout/example and resources/ui_layout/default it starts fine and seems to work perfectly.

serproqnx, 2022-07-18:
same, arch

GrinTechwa, 2022-07-18:
> Looks like SS on the arch repo finally updated today. Has anyone tried it?

Sent from my Verizon, Samsung Galaxy smartphone

serproqnx, 2022-07-18:
prusa-slicer-2.4.2-6-x86_64 segfault on arch too

Darkstar77, 2022-07-21:
Same over here, Manjaro Linux. I am unable to rollback, though I did try. Definitely an issue with the new wx library.

serproqnx, 2022-07-22:
> 

prusa works now after last core updates just sudo pacman -Suy
superslicer still segfault

serproqnx, 2022-07-22:
maybe it's related https://github.com/prusa3d/PrusaSlicer/commit/ec6192a

serproqnx, 2022-07-23:
```
[serp@dsktp ~]$ prusa-slicer -loglevel 5
[2022-07-23 09:45:07.641084] [0x00007f6c24c3a840] [debug]   full path: 7299071426822281241
[2022-07-23 09:45:07.641100] [0x00007f6c24c3a840] [debug]   single instance: undefined. other params: prusa-slicer;-loglevel;5
[2022-07-23 09:45:07.641104] [0x00007f6c24c3a840] [debug]   full lock path: /home/serp/.config/PrusaSlicer/cache/7299071426822281241.lock
[2022-07-23 09:45:07.641238] [0x00007f6c24c3a840] [debug]   Creating lockfile.
[2022-07-23 09:45:07.641244] [0x00007f6c24c3a840] [info]    Instance check: Another instance not found or single-instance not set.
[2022-07-23 09:45:07.677301] [0x00007f6c24c3a840] [trace]   System language detected (user locales and such): en_US
[2022-07-23 09:45:07.745584] [0x00007f6c24c3a840] [trace]   Best translation language detected (may be different from user locales): en_US
[2022-07-23 09:45:07.745636] [0x00007f6c24c3a840] [trace]   Switching wxLocales to en_US
[2022-07-23 09:45:07.846819] [0x00007f6c24c3a840] [info]    Checking if indices need to be installed from resources...
```
```
[serp@dsktp ~]$ superslicer -loglevel 5
[2022-07-23 09:47:06.783316] [0x00007f3f3a1cbd80] [debug]   full path: 16617057003357181261
[2022-07-23 09:47:06.783345] [0x00007f3f3a1cbd80] [debug]   single instance: undefined. other params: superslicer;-loglevel;5
[2022-07-23 09:47:06.783349] [0x00007f3f3a1cbd80] [debug]   full lock path: /home/serp/.config/SuperSlicer/cache/16617057003357181261.lock
[2022-07-23 09:47:06.783378] [0x00007f3f3a1cbd80] [debug]   Creating lockfile.
[2022-07-23 09:47:06.783382] [0x00007f3f3a1cbd80] [info]    Instance check: Another instance not found or single-instance not set.
[2022-07-23 09:47:06.815074] [0x00007f3f3a1cbd80] [trace]   System language detected (user locales and such): en_US
Segmentation fault (core dumped)

```

supermerill, 2022-07-24:
@serproqnx can you try latest nightly? i merged the commit you linked.

veehexx, 2022-07-24:
i've just tried this myself (Fedora36):

> $ SuperSlicer-gtk3.AppImage-nightly137/SuperSlicer_ubu64.AppImage --loglevel 5
> [2022-07-24 13:49:47.570853] [0x00007f0d73020540] [trace]   Initializing StaticPrintConfigs
> SuperSlicer GUI initialization failed: boost::filesystem::copy_file: **File exists:** "/tmp/.mount_SuperSUKqZFE/bin/../resources/ui_layout/example/colors.ini", "/home/USER/.config/SuperSlicer/ui_layout/example/colors.ini"
> 
> (process:53507): Gtk-CRITICAL **: 13:49:47.769: _gtk_css_lookup_resolve: assertion '(((__extension__ ({ GTypeInstance *__inst = (GTypeInstance*) ((provider)); GType __t = ((_gtk_style_provider_private_get_type ())); gboolean __r; if (!__inst) __r = (0); else if (__inst->g_class && __inst->g_class->g_type == __t) __r = (!(0)); else __r = g_type_check_instance_is_a (__inst, __t); __r; }))))' failed
> 
> (process:53507): GLib-GObject-CRITICAL **: 13:49:47.769: g_object_set_data_full: assertion 'G_IS_OBJECT (object)' failed
> 
> (process:53507): Gtk-ERROR **: 13:49:47.769: Can't create a GtkStyleContext without a display connection
> Trace/breakpoint trap (core dumped)

however, if i rename the ~/.config/SuperSlicer/ folder for a first-time run, then the log changes slightly:

> $ SuperSlicer-gtk3.AppImage-nightly137/SuperSlicer_ubu64.AppImage --loglevel 5
> [2022-07-24 13:53:23.101913] [0x00007fd2f8290540] [trace]   Initializing StaticPrintConfigs
> SuperSlicer GUI initialization failed: boost::filesystem::copy_file: **Invalid cross-device link:** "/tmp/.mount_SuperSRRnwKV/bin/../resources/ui_layout/example/colors.ini", "/home/USER/.config/SuperSlicer/ui_layout/example/colors.ini"
> 
> (process:54053): Gtk-CRITICAL **: 13:53:23.303: _gtk_css_lookup_resolve: assertion '(((__extension__ ({ GTypeInstance *__inst = (GTypeInstance*) ((provider)); GType __t = ((_gtk_style_provider_private_get_type ())); gboolean __r; if (!__inst) __r = (0); else if (__inst->g_class && __inst->g_class->g_type == __t) __r = (!(0)); else __r = g_type_check_instance_is_a (__inst, __t); __r; }))))' failed
> 
> (process:54053): GLib-GObject-CRITICAL **: 13:53:23.303: g_object_set_data_full: assertion 'G_IS_OBJECT (object)' failed
> 
> (process:54053): Gtk-ERROR **: 13:53:23.303: Can't create a GtkStyleContext without a display connection
> Trace/breakpoint trap (core dumped)

Note that the second line changes from 'file exists' to 'Invalid cross-device link'. To my untrained eye, it looks like something going wrong with colors.ini update/creation and it having an effect on the GUI/Display initialisation process?

zap117, 2022-07-24:
same here on nightly, using gnome 42.3 and arch based

supermerill, 2022-07-24:
> however, if i rename the ~/.config/SuperSlicer/ folder for a first-time run

rename from what? It wasn't created?

veehexx, 2022-07-24:
> rename from what? It wasn't created?

rename to anything (or delete it); so when you start SuperSlicer, a new '~/.config/SuperSlicer/' folder will be created.

zap117, 2022-07-24:
rename and delete don't work. even tried with a fresh OS install and the same fault happens to SS

veehexx, 2022-07-24:
> rename and delete don't work. even tried with a fresh OS install and the same fault happens to SS

I wasn't saying that would fix it, but in my post 5 hours ago with 2 different outputs; one was with an existing '~/.config/SuperSlicer/' folder, and the other was with that folder deleted/renamed (so next run of SuperSlicer would recreate them).

what i was getting at was based on the original report/actual output of my computer, wxwidgets does not seem to be the case and is appears to be something around file creation/update/colors.ini.

hopefully thats clearer :)

@supermerill - your earlier commit build (257157) works for me - thanks!
I've deleted ~/.config/superslicer/ for a clean/fresh run, so incase you want to do some more fixes with the 'boost::filesystem::permisions' messages, or there's something else that that isnt showing during startup, here is my current output:

> $ SuperSlicer-gtk3.AppImage-nightly139/SuperSlicer_ubu64.AppImage --loglevel=5
> [2022-07-24 19:44:01.218396] [0x00007fd863f06540] [trace]   Initializing StaticPrintConfigs
> [2022-07-24 19:44:01.370686] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.372591] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.374202] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.377242] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.378806] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.380374] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.383094] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.384863] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.387036] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.388715] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.390399] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.391914] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.394550] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.396228] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.397953] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.399662] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.401344] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.404604] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.406481] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.408300] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.409961] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
> [2022-07-24 19:44:01.413224] [0x00007fd863f06540] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
[... cut ...]
> [2022-07-24 19:44:06.250782] [0x00007fd863f06540] [trace]   Vendor bundle: `/tmp/.mount_SuperSVr5ABE/bin/../resources/profiles/Zonestar.ini`: 17 profiles loaded.
> [2022-07-24 19:44:06.278707] [0x00007fd863f06540] [trace]   Vendor bundle: `/tmp/.mount_SuperSVr5ABE/bin/../resources/profiles/gCreate.ini`: 45 profiles loaded.
> [2022-07-24 19:44:07.491417] [0x00007fd863f06540] [debug]   BuildVolume bed_shape clasified as: Rectangle
> [2022-07-24 19:44:07.549450] [0x00007fd863f06540] [info]    Running ConfigWizard, reason: 0, start_page: 0
> [2022-07-24 19:44:07.549508] [0x00007fd863f06540] [debug]   Desktop integration desktop file path:

veehexx, 2022-07-24:
Second run does not exhibit the same errors, so i think we're good?

> $ SuperSlicer-gtk3.AppImage-nightly139/SuperSlicer_ubu64.AppImage --loglevel=5
> [2022-07-24 19:54:41.026871] [0x00007f8d4ed75540] [trace]   Initializing StaticPrintConfigs
> [2022-07-24 19:54:41.197080] [0x00007f8d4ed75540] [debug]   full path: 9025485289904292181
> [2022-07-24 19:54:41.197108] [0x00007f8d4ed75540] [debug]   single instance: undefined. other params: /tmp/.mount_SuperS7tUJU1/bin/superslicer;--loglevel=5
> [2022-07-24 19:54:41.197117] [0x00007f8d4ed75540] [debug]   full lock path: /home/USER/.config/SuperSlicer/cache/9025485289904292181.lock
> [2022-07-24 19:54:41.197218] [0x00007f8d4ed75540] [debug]   Creating lockfile.
> [2022-07-24 19:54:41.197228] [0x00007f8d4ed75540] [info]    Instance check: Another instance not found or single-instance not set.
> [2022-07-24 19:54:41.301594] [0x00007f8d4ed75540] [trace]   System language detected (user locales and such): en_GB
> [2022-07-24 19:54:41.305505] [0x00007f8d4ed75540] [trace]   Best translation language detected (may be different from user locales): en_GB
> [2022-07-24 19:54:41.305550] [0x00007f8d4ed75540] [trace]   Switching wxLocales to en_GB
> [2022-07-24 19:54:44.063560] [0x00007f8d4ed75540] [info]    Checking if indices need to be installed from resources...
> 
> (superslicer:84482): Gtk-CRITICAL **: 19:54:44.102: gtk_window_resize: assertion 'height > 0' failed
> [2022-07-24 19:54:44.592742] [0x00007f8d4ed75540] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 125MB; Shared memory: 91MB; Private memory: 34MB; Peak memory usage: 125MB
> [2022-07-24 19:54:44.593656] [0x00007f8cf27fc640] [trace]   Dbus object /org/slic3r/superslicer/InstanceCheck/Object9025485289904292181 registered. Starting listening for messages.
> [2022-07-24 19:54:44.664871] [0x00007f8d4ed75540] [warning] Warning, can't find file script '/home/USER/.config/SuperSlicer/ui_layout/default/filament.as', is it needed? Even if something need it, it won't execute.
> [2022-07-24 19:54:44.698634] [0x00007f8d4ed75540] [warning] Warning, can't find file script '/home/USER/.config/SuperSlicer/ui_layout/default/sla_material.as', is it needed? Even if something need it, it won't execute.
> [2022-07-24 19:54:44.711377] [0x00007f8d4ed75540] [warning] Warning, can't find file script '/home/USER/.config/SuperSlicer/ui_layout/default/printer.as', is it needed? Even if something need it, it won't execute.
> [2022-07-24 19:54:44.842092] [0x00007f8d4ed75540] [debug]   Set field from key s_support_fff as int 0 modified to 0
> [2022-07-24 19:54:44.860419] [0x00007f8d4ed75540] [debug]   BuildVolume bed_shape clasified as: Rectangle
> [2022-07-24 19:54:45.544603] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/Anycubic.ini`: 103 profiles loaded.
> [2022-07-24 19:54:45.562381] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/Artillery.ini`: 18 profiles loaded.
> [2022-07-24 19:54:45.588939] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/BIBO.ini`: 33 profiles loaded.
> [2022-07-24 19:54:45.649864] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/Basic.ini`: 80 profiles loaded.
> [2022-07-24 19:54:45.728965] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/Creality.ini`: 151 profiles loaded.
> [2022-07-24 19:54:45.752489] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/FLSun.ini`: 24 profiles loaded.
> [2022-07-24 19:54:45.764066] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/INAT.ini`: 23 profiles loaded.
> [2022-07-24 19:54:45.774972] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/Jubilee.ini`: 13 profiles loaded.
> [2022-07-24 19:54:45.781754] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/LulzBot.ini`: 5 profiles loaded.
> [2022-07-24 19:54:45.801145] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/MakerGear.ini`: 26 profiles loaded.
> [2022-07-24 19:54:45.819397] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/Meltingplot.ini`: 27 profiles loaded.
> [2022-07-24 19:54:46.033603] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/PrusaResearch.ini`: 809 profiles loaded.
> [2022-07-24 19:54:46.061816] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/RatRig.ini`: 16 profiles loaded.
> [2022-07-24 19:54:46.079228] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/Snapmaker.ini`: 16 profiles loaded.
> [2022-07-24 19:54:46.128353] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/TriLAB.ini`: 107 profiles loaded.
> [2022-07-24 19:54:46.140236] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/Ultimaker.ini`: 8 profiles loaded.
> [2022-07-24 19:54:46.210722] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/Vertex.ini`: 171 profiles loaded.
[... cut ...]
> [2022-07-24 19:54:46.366316] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/Zonestar.ini`: 17 profiles loaded.
> [2022-07-24 19:54:46.392455] [0x00007f8d4ed75540] [trace]   Vendor bundle: `/tmp/.mount_SuperS7tUJU1/bin/../resources/profiles/gCreate.ini`: 45 profiles loaded.
> [2022-07-24 19:54:47.650132] [0x00007f8d4ed75540] [debug]   BuildVolume bed_shape clasified as: Rectangle
> [2022-07-24 19:54:47.720989] [0x00007f8d4ed75540] [info]    Running ConfigWizard, reason: 0, start_page: 0
> [2022-07-24 19:54:47.721045] [0x00007f8d4ed75540] [debug]   Desktop integration desktop file path:
>

supermerill, 2022-07-24:
I created a nightly (commit 487e5c8) where i changed the boost::copy_file by the more robust one that prusa has created for "when copying on some exotic file system, for example Linux on Chrome."

dusakus, 2022-07-24:
> > 
> 
> prusa works now after last core updates just sudo pacman -Suy superslicer still segfault

more likely related to a change in the buildscript, they now build prusa's patched wxWidgets
https://github.com/archlinux/svntogit-community/commit/143531d44a95acb3e84a64d5ab6f2e23ac8216a1

gyordanov, 2022-07-24:
i can confirm that "test for #2938" build does not segfault for me anymore on arch

distinctjuggle, 2022-07-26:
So not trying to be, "that guy" or anything, but when could we expect these nightly changes to be added to a main release?

supermerill, 2022-07-28:
update: there is a big bug. I'll delay the release for a day or two to try to catch it.

sdooweloc, 2022-07-28:
> update: there is a big bug. I'll delay the release for a day or two to try to catch it.

I'm glad you were able to track it down! Thank you for the work you do!

olby, 2022-07-29:
This also fix #2936

closed 2024-02-11 by supermerill