supermerill/SuperSlicer #2940: Mosaic: failed in wxBitmap

foreachthing, 2022-07-15:
### What happened?

Any .png I've tried, all fail with the same message:

[image]

### Project file & How to reproduce

Generate -> Mosaic from picture -> File... -> [OK] leads to crash.

### Version

2.4.58.3_win64_220714

### Operating system

Win106x4

### Printer model

Ultimaker 2

supermerill, 2022-07-15:
can you give me your png?
mine are working.

foreachthing, 2022-07-15:
Um ... trying different png at home now.
Do you have a size restriction? I got on 32x32 but it just shows white.

Any .png with transparency will crash.

Ok! I got this image 
[image]
working with these settings: 
[image]

I didn't pay close enough attention to the settings - there was an offset still in place.
Also, on the first try, I've lost the generate button in the lower left corner.

Now I got to boost contrast so I won't need 102 extruders 😏

closed 2022-07-15 by foreachthing

supermerill, 2022-07-15:
in the second tab, you can define the acceptable colors (your spools).
Then, you can choose the function to merge colors.