supermerill/SuperSlicer #2950: SuperSlicer 2.4.58.3 crashes on startup on macOS 12.4

nielsvz, 2022-07-16:
### What happened?

SuperSlicer 2.4.58.3 crashes on startup on macOS 12.4

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [43591]
Path:                  /Users/USER/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.3_2022-07-13 (SuperSlicer_2.4.58.3_2022-07-13)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-07-16 18:19:26.0253 +0200
OS Version:            macOS 12.4 (21F79)
Report Version:        12
Bridge OS Version:     6.5 (19P5071)
Anonymous UUID:        C3DE8E7A-C2D0-40A8-3A9D-385D3EB773EB

Sleep/Wake UUID:       7F252B48-5883-44E8-BD3F-B3BAB2820A52

Time Awake Since Boot: 530000 seconds
Time Since Wake:       656 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: /Users/USER/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	       0x112f970be __abort_with_payload + 10
1   dyld                          	       0x112fad1d2 abort_with_payload_wrapper_internal + 80
2   dyld                          	       0x112fad204 abort_with_payload + 9
3   dyld                          	       0x112f5b76f dyld4::halt(char const*) + 375
4   dyld                          	       0x112f575a5 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 3995
5   dyld                          	       0x112f564d4 start + 388

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7bbce9658  rdx: 0x00007ff7bbce9ac0
  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7bbce96a0  rsp: 0x00007ff7bbce9658
   r8: 0x00007ff7bbce96c0   r9: 0x0000000000000000  r10: 0x0000000000000081  r11: 0x0000000000000246
  r12: 0x0000000000000081  r13: 0x00007ff7bbce9ac0  r14: 0x0000000000000001  r15: 0x0000000000000006
  rip: 0x0000000112f970be  rfl: 0x0000000000000246  cr2: 0x0000000112f5de74
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x112f51000 -        0x112fbcfff dyld (*) <b70ce1ec-b902-3852-8268-05de00bfa8d5> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
[... 102 lines cut ...]
      "rolloutId" : "5ffde50ce2aacd000d47a95f",
      "factorPackIds" : {

      },
      "deploymentId" : 240000167
    }
  ],
  "experiments" : [

  ]
}
}
```

### Project file & How to reproduce

Start SuperSlicer

### Version

2.4.58.3

### Operating system

macOS 12.4

nielsvz, 2022-07-16:
SuperSlicer will start normally if I install the `zstd` package through Homebrew.

ccfman2004, 2022-07-17:
> SuperSlicer will start normally if I install the `zstd` package through Homebrew.

This does fix the issue with it crashing.

Diramu, 2022-07-27:
Same problem with macOS 11.6.7.
After running 'brew install zstd'
Works fine (well it started...)

supermerill, 2022-07-27:
you can try with this:
https://mega.nz/file/01BAVLyJ#D36IjInHUg5_l-uW-x4XY7BRhsQS4jLioeDR5517Shs
shouldn't need anything else on an intel Mac

first test for an universal app