supermerill/SuperSlicer #2951: Superslicer does not slice part of stl object

adriaan-vd-graaf, 2022-07-17:
### What happened?

Hi, 

First off, thank you for building and maintaining SuperSlicer, it's a great piece of software.
Unfortunately I have an issue that I haven't found in the list of issues. 

I have a part that does not get sliced correctly. Part of the object is missing when sliced but it remains in the 3d editor view. I'm attaching a 3mf file to show you the part.
there are three parts in the 3mf, one combined, and two parts that are cut through the middle to remove any supports, which are supposed to be glued together after printing.

The issue is in the part called '000_buis_connector._bottom_stl'. 
As you can see, the top section does not get sliced.

Thank you for your time and I sincerely hope this is user error :-)

best wishes, 
Adriaan

### Project file & How to reproduce

[2022_07_17_superslicer_issue.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9128397/2022_07_17_superslicer_issue.3mf.zip)

### Version

2.4.58.2

### Operating system

MACOS 12.4

### Printer model

Voron 2.4 with afterburner (not important)

neophyl, 2022-07-18:
If you look at both of your partial models you will see the yellow warning triangle.  This indicates issues with your models.  The top has 68 open edges and the bottom has 48 open edges.
If you fix the models then they slice correctly.

[2022_07_17_superslicer_issue_fixed.zip](https://github.com/supermerill/SuperSlicer/files/9130870/2022_07_17_superslicer_issue_fixed.zip)

adriaan-vd-graaf, 2022-07-18:
Thanks for the reply, I agree that there is an issue with the partial models. 
These partial models have been created by the 'cut' tool of SuperSlicer on a model that does not have these issues. Perhaps this wasn't clear in the initial bug report so I guess this is the bug that I'm referring to.

neophyl, 2022-07-18:
Yes the cut function on this model is producing errors.  If you leave it exactly half way.  Changing the cut plane up or down slightly results in models without the open edge issues that slice correctly.

I suspect the cut is passing mathematically through several vertices and its doesnt know how to cut, normally a vert is created on each edge to give 2, 1 for each new object.  I think its having trouble if one is already in the same position.  

The same problem occurs with this model when Cut with Prusa Slicer so its a common issue with the source project too.

adriaan-vd-graaf, 2022-07-18:
Thanks for the information and the workaround!
It's too bad I designed something so terrible :-P 
I'll close the issue now.

best wishes, Adriaan

neophyl, 2022-07-18:
I wouldn't close it, the model itself is perfect :)  There are no inconsistencies with the original.

Personally what I would be tempted to do is open an issue on the PS github with a saved PS project (if there isnt one already on there).  They have way more resources.  If they already have an open issue then adding your model as another test one they can use to verify a fix is always useful too.