supermerill/SuperSlicer #2953: print comes out 5mm longer than model

billsygtb, 2022-07-18:
### What happened?

Plated up a long thin model which is  237mm long and when printed it comes out 242mm in length.

exported the file from fusion 360 twice (in STL and 3MF) and tried several times with and without brim etc with the same result.
the object reports the correct size in the properties window when at 90 degrees, but i cant tell once rotated to fit the bed,

I test printed the same 3MF with Cura and it printed properly. 

I am printing with a .6 nozzle, object rotated to fit the bed and 

I dont get the same results with smaller prints.

### Project file & How to reproduce

[erics brake light prints long x 5mm.zip](https://github.com/supermerill/SuperSlicer/files/9131137/erics.brake.light.prints.long.x.5mm.zip)

### Version

2.4.58.3 also happened on 2.4.58.2

### Operating system

Windows 10 20H2

### Printer model

Ender 3, BTT SKR Mini E3 board

neophyl, 2022-07-18:
Well your model on the build plater is still 237mm long when rotated at 45 degrees as placed.  You can select Local Coordinates instead of World to check that.

Your filament profile which is set to Nylon type (rather than ABS which seems to be the base profile type)  and DOES have a shrinkage value of 97.5% configured.  That is going to apply a 2.5% scaling to combat the defined shrinkage.  That works out to 242.925mm for a 237mm long object, so if you are getting an extra 5mm I would assume its coming from your shrinkage adjustment.  In effect you have said your filament shrinks 2.5% when printed so the slicer is adding that on to the gcode generated.  The default ABS filament profile normally has 100% in there, ie no shrinkage so no compensation.  So its doing exactly what you have it configured to do.

billsygtb, 2022-07-18:
I reckon your spot on, you bloody legend :) 
Ive been staring at the settings for 2 days and missed that the entire time as i only looked at what id changed.

how that setting changed i dont know, i was sure I based this profile off of my PETG profile and never changed that setting. 
ill run a print now its reset and close this once confirmed.

LAP87, 2022-07-26:
I had the reverse issue once, printing a parabola shaped object (fender for a onewheel), between the outermost tips is was lacking 10mm, making the 3-day print useless 👎🏻 
Both Cura and SS/PS let me down with that one, Ideamaker slicer finally let me print it correctly, which was lucky for me!