supermerill/SuperSlicer #2955: SuperSlicer 2.4.58.3 no infill over bridge

aragor88, 2022-07-18:
### What happened?

On the layer where is bridging and last solid bottom layer, there is no infill over bridge.

### Project file & How to reproduce

[longplank2.zip](https://github.com/supermerill/SuperSlicer/files/9134535/longplank2.zip)
[image]
[image]

### Version

2.4.58.3

### Operating system

Windows 10

### Printer model

MK3S+

supermerill, 2022-07-21:
set solid_over_perimeter (no solid infill over X perimeters) to 0

TaffoFox, 2022-07-29:
I have the same issue. Set it to 0 for now.

closed 2024-02-11 by supermerill