supermerill/SuperSlicer #2956: Crash on mac m1

Tenttu78, 2022-07-19:
### What happened?

I have installed some beta version of SuperSlicer_2.4.58.3 and it works fine, but when i try use this new version or any other versions it crashes. Do i need to uninstall something?

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [27681]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.3_2022-07-13 (SuperSlicer_2.4.58.3_2022-07-13)
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-07-19 09:43:21.9968 +0300
OS Version:            macOS 12.4 (21F79)
Report Version:        12
Anonymous UUID:        2396307F-890D-6C1E-ADD3-E2E04CE52B69

Time Awake Since Boot: 170000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   Rosetta Runtime Routines	       0x10b4cf940 ???
1   <translation info unavailable>	       0x10b51906c ???
2   dyld                          	       0x207dc21d2 abort_with_payload_wrapper_internal + 80
3   dyld                          	       0x207dc2204 abort_with_payload + 9
4   dyld                          	       0x207d7076f dyld4::halt(char const*) + 375
5   dyld                          	       0x207d6c5a5 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 3995
6   dyld                          	       0x207d6b4d4 start + 388

Thread 1:: com.apple.rosetta.exceptionserver
0   runtime                       	    0x7ff7ffd9e9c4 0x7ff7ffd9a000 + 18884
1   runtime                       	    0x7ff7ffdac460 0x7ff7ffd9a000 + 74848
2   runtime                       	    0x7ff7ffdaded8 0x7ff7ffd9a000 + 81624

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000006   x1: 0x0000000000000001   x2: 0x00000003106a8ac0   x3: 0x0000000000000075
    x4: 0x00000003106a86c0   x5: 0x0000000000000000   x6: 0xd57941c4f2d600b4   x7: 0x00000020106a86d0
    x8: 0x00000003106a86c0   x9: 0x0000000000000000  x10: 0x0000000000000075  x11: 0x00000003106a86c0
   x12: 0x0000000000000075  x13: 0x00000003106a8ac0  x14: 0x0000000000000001  x15: 0x0000000000000006
   x16: 0x0000000000000209  x17: 0x0000000000000000  x18: 0x8000000113538b70  x19: 0x000000010b4c5000
   x20: 0x00000000000007ff  x21: 0x000000011353dfa0  x22: 0xffffffffffffffff  x23: 0x000000010b4cf914
   x24: 0x0000000207dc21d2  x25: 0x000000010b4cf1f4  x26: 0x000000010b517708  x27: 0x0000000000000000
   x28: 0x0000000207dac0be   fp: 0x000000002c000000   lr: 0x000000010b51906c
    sp: 0x0000000113538b70   pc: 0x000000010b4cf940 cpsr: 0x80001000
   far: 0x000000010b519000  esr: 0x56000080  Address size fault

Binary Images:
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???
       0x207d66000 -        0x207dd1fff dyld (*) <b70ce1ec-b902-3852-8268-05de00bfa8d5> /usr/lib/dyld
    0x7ff7ffd9a000 -     0x7ff7ffdc9fff runtime (*) <0ee8ce32-639c-316a-9551-dab9a2e0c8a5> /usr/libexec/rosetta/runtime

External Modification Summary:
  Calls made by other processes targeting this process:
[... 154 lines cut ...]
USB Device: USB Receiver
USB Device: USB30Bus
USB Device: Keyboard Hub
USB Device: Apple Keyboard
USB Device: USB 2.0 Hub
USB Device: USB 2.0 Hub
USB Device: U24XL
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.

### Project file & How to reproduce

No file.

### Version

2.4.58.3

### Operating system

macos 12.4 (21F79)

### Printer model

voron 2.4

supermerill, 2022-07-19:
Seems like you now need to install zstd, even on mac arm.
install homebrew
then type "brew install zstd" in a terminal.

Tenttu78, 2022-07-20:
That didnt help.

supermerill, 2022-07-20:
you still have 
> Library not loaded: /usr/local/opt/**zstd**/lib/libzstd.1.dylib
in your crash report?

Tenttu78, 2022-07-20:
Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file)
(terminated at launch; ignore backtrace)

supermerill, 2022-07-20:
what are the results of the command `brew --prefix zstd` ?
and `ls -al /usr/local/opt/zstd/lib` ?

Tenttu78, 2022-07-20:
Last login: Wed Jul 20 08:38:58 on ttys000
tenhopellikka@Tenho-Mac-mini ~ % brew --prefix zstd
/opt/homebrew/opt/zstd
tenhopellikka@Tenho-Mac-mini ~ % ls -al /usr/local/opt/zstd/lib
ls: /usr/local/opt/zstd/lib: No such file or directory
tenhopellikka@Tenho-Mac-mini ~ %

supermerill, 2022-07-20:
are you using the intel version on a m1 mac?

Tenttu78, 2022-07-20:
I just downloaded arm version and now it works. I am so sorry.

closed 2022-07-20 by supermerill