supermerill/SuperSlicer #2961: No Filament Control

SphaeroX, 2022-07-21:
### What happened?

I wanted to set the filament cooling for the PVA to 0% with my dual extruder. But unfortunately the settings are not applied during slicing.

It only works if I do it manually via the filament GCode....

### Project file & How to reproduce

[image]
[image]

[no_fan_control.zip](https://github.com/supermerill/SuperSlicer/files/9158127/no_fan_control.zip)

### Version

Version 2.4

### Operating system

windows 11

### Printer model

Other

supermerill, 2022-07-23:
Currently, the cooling is a post-process script that works on the generated gcode.
That's because the cooling buffer don't recognize `T1` as a toolchange command, as the toolchange command for klipper is 
`ACTIVATE_EXTRUDER EXTRUDER=1`
So it can't detect the extruder change, and so can't change it.

supermerill, 2022-07-23:
won't rely ont he gcode only for next version.

themacboy, 2022-08-11:
Good morning, Im not sure what that commit does. I look at the genereted code for my idex printer and I dont see any diference, in gcode. Probalby I missunderstood something, my english is very poor.

themacboy, 2022-08-11:
Personally I control all that in klipper using a personal macro that I activate in every tool_change_gcode (i think all people do that way).
 But not sure if this commits improve something about this.

closed 2024-02-11 by supermerill