supermerill/SuperSlicer #2962: SuperSlicer 2.4.58.3 does not run on macOS Catalina 10.15.7

rainell, 2022-07-21:
### What happened?

SuperSlicer 2.4.58.3 does not run on macOS Catalina 10.15.7. I had to go back to SuperSlicer 2.3.57.12
[image]

### Project file & How to reproduce

SuperSlicer 2.4.58.3 does not run on macOS Catalina 10.15.7

### Version

2.4.58.3

### Operating system

macOS Catalina 10.15.7

jwdanie, 2022-07-25:
Same crash on a fully current Monterey 12.4 (21F79) system as well.
Had to revert back to previous release.

supermerill, 2022-07-25:
@jwdanie max intel with intel version of Susi and zstd installed?

jwdanie, 2022-07-25:
I have just been using the .dmg as given up to this point. Yes, an Intel iMac. No idea about Susi or zstd as mentioned, I just use the .dmg file as is.

supermerill, 2022-07-26:
SuSi = SuperSlicer.

zstd: see the release post, or my answer above.
```
You are on an intel macbook, right?
If yes, this version need you to install manually zsdt. I don't know how to embed it in the dmg.

The simplest way is to install [hombrew](https://brew.sh/) and then type brew install zstd in the command terminal.
```

jwdanie, 2022-07-26:
Sorry for the SuSi confusion!
I'll just stay with 2.3.57.12 then which is working fine.

supermerill, 2022-07-26:
you can try with this:
https://mega.nz/file/01BAVLyJ#D36IjInHUg5_l-uW-x4XY7BRhsQS4jLioeDR5517Shs
shouldn't need anything else on an intel Mac

soldop, 2022-07-31:
Thanks @supermerill that link did the trick for me as well.

closed 2022-11-07 by rainell