supermerill/SuperSlicer #2964: "No Sparse Layers" Switching to Second Color in Air

joey2point0, 2022-07-22:
### Version
2.4.58.3
### Operating system type + version
Windows 10 Build 19044

### Behavior
"No Sparse Layers (EXPERIMENTAL)" setting enabled and using multilateral print settings. Generated Gcode wants to purge the second color in the air, instead of dropping the extruder back down to build plate. I've done this exact print successfully in 2.3.58.2 and in 2.3.57.12
[image]

supermerill, 2022-07-22:
can I have your example project?

joey2point0, 2022-07-23:
[Plants Names - Genvani.zip](https://github.com/supermerill/SuperSlicer/files/9174392/Plants.Names.-.Genvani.zip)

supermerill, 2022-07-23:
Seems like it's the way intended by prusa.

ibrahimcengiz, 2022-08-12:
Not fixed on 2.4.58.4 relase

spikeygg, 2022-09-04:
I have this problem too, it's pretty easy to see. It worked in 2.3.57 but broke by 2.4.58 and still persists in 2.5.59. You can take a simple print and add an extruder change half-way up the print with sparse layers on. When you slice it, it will do the purge in mid-air half-way up the print. It looks like the new version doesn't do the 1st extruder purge cycle, maybe that's related?

Here's an example of the same print in both 2.5.59 (left) and 2.3.57 (right) for comparison:
[image]

mann1861, 2022-11-07:
I'm also experiencing this in 2.5.59. Photos attached shows the wipe tower layers aren't dropping down to the bed after the empty layers have been deleted:
[image]

cryoglobe, 2022-11-15:
I confirm it's still there in 2.4.59 , had to take back the 2.4.52 to slice in such a case

Kermit647, 2024-04-22:
2.5.59.8 still not fixed