supermerill/SuperSlicer #2965: Vase Mode: Extrusion width Inconsistent

DanielDC88, 2022-07-22:
### Version
2.3.57.12

### Operating system type + version
Windows 10, latest 

### Behavior
* _Describe the problem_
Extrusion width changes on certain sections of part sliced in vase mode. Same config loaded into Prusaslicer does not have this issue
* _Steps needed to reproduce the problem_
Load part, enable vase mode, set external perimeter extrusion width to 0.99 mm
* _Expected Results_
* all lines are 0.99mm
* _Actual Results_
a small section is at a lower line width (smaller than nozzle)
 * _Screenshots if useful_
[image]

#### Project File (.3MF) where problem occur
[Vase Extrusion Width PrusaSlicer.zip](https://github.com/supermerill/SuperSlicer/files/9169922/Vase.Extrusion.Width.PrusaSlicer.zip)
[Vase Extrusion Width SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/9169923/Vase.Extrusion.Width.SuperSlicer.zip)

neophyl, 2022-07-22:
I don't know exactly why its doing it in that area only but I do know how to stop it doing it.

Under Perimeters & Shell Set your overhangs Thresholds for bridge speed and fan to something like 200% for both.  (arbitrary big number that I tried first).  That stops the detection of any of the perimeters being overhang.  When you do that it applies your 0.99mm width all around.
Why its only doing it to that small area instead of everywhere else that its detecting as overhang I dont know.  One for Merill I guess.

But at least that should let you slice your object.
There's no issue with PS as that doesn't have those settings.  The thing you have to remember is while similar they aren't the same :)

supermerill, 2022-07-26:
Prusaslicer has a hardcoded 50% nozzle diameter threshold for overhangs.
Superslicer has a configurable amount, that can be a % of the perimeter extrusion width (your is set to 75%).

you have a 1mm extrusion width for a 0.4mm nozzle. that's why there is a big difference.