supermerill/SuperSlicer #2968: Recomended min wall thickness calculation

cfytrok, 2022-07-23:
1. The recommended min wall thickness does not take into account width and spacing settings.
2. It doesn't show results for odd number of perimeters, like Prusa Slicer. Even if check "Thin walls".
3. Results are different with Prusa slicer. Who is right?
[image]
[image]

supermerill, 2022-07-23:
> The recommended min wall thickness does not take into account width and spacing settings.

it is, just checked.

> It doesn't show results for odd number of perimeters, like Prusa Slicer. Even if check "Thin walls"

it show the result up to the amont of periemter loop you set above.

> Results are different with Prusa slicer. Who is right?

I'm right, I spend a good amount of time to refine it.

pedroloco2, 2022-07-23:
I did several tests and I think cfytrok recommendation may be correct. I clarify that I have not tried with thin wall.

cfytrok, 2022-07-23:
> it is, just checked.

Sorry, it seems changing esternal perimeter width changes recomended thickness. But I changed default width and set other width to 0. According to the hint, external perimeter should use default width. And gcode preview confirms it. But Recomended thickness doesn't change.
[image]
[image]

>it show the result up to the amont of periemter loop you set above.

Prusa slicer shows min thickness per line instead of per loop. It's more accurate.

supermerill, 2022-07-23:
> Prusa slicer shows min thickness per line instead of per loop. It's more accurate. 1 loop=2 lines

you can't have an odd number of perimeter "lines" anyway in Prusaslicer, so there is no point.

> But I changed default

Right, it's not updated if using the default width. will fix.
Seems like the spacing text isn't updated as well when the default change.

closed 2024-02-11 by supermerill