supermerill/SuperSlicer #2969: FreeCAD python script on linux

mojamuto, 2022-07-23:
### What happened?

I have FreeCAD installed, when I generate a script it tells me : 
"Error, freecad python (version 0.19 or higher) is not found in 'bin/python', please update the freecad python path in preferences".

[image]

Check where FreeCAD is installed:

[image]

I type the path in FreeCAD preferences path: /usr/lib/freecad.

[image]

and it tells me this
Error, freecad python (version 0.19 or higher) is not found in '/usr/lib/freecad/bin/python', please update freecad python path in preferences.

[image]

What am I doing wrong?

### Project file & How to reproduce

...

### Version

2.4.58.3

### Operating system

void linux

### Printer model

prusa bear

supermerill, 2022-07-23:
I don't remeber if it was really the case, but freecad can come with a python install.
I check, and there isn't one in the 0.19 on ubuntu22.04.
I'll change the software to revert to python3 system python if it can't find any freecad python.

In the meantime, you can create a symlink from `/usr/lib/freecad/bin/python` to your python3

mojamuto, 2022-07-24:
that worked, thank you very much

closed 2022-07-24 by mojamuto