supermerill/SuperSlicer #2974: Skip Retraction when Infilling

zjagoo, 2022-07-23:
Hello,

One feature that would be useful would be to skip retraction when infilling. This is specially useful for printers with bowden configuration.
The intended behavior would be to still retract (if enabled) during solid layers but no retraction during infilling since the blobs would be hidden.
Thanks

tome9111991, 2022-07-24:
you try only Retract when crossing perimeter?

ElliottBridge, 2022-07-25:
> Hello,
>
> One feature that would be useful would be to skip retraction when infilling. This is specially useful for printers with bowden configuration.
> The intended behavior would be to still retract [...]

I'm guessing this is so you can improve reliability with flexible filaments?