supermerill/SuperSlicer #2975: Allow - Only one perimeter - setting to have an effect over multiple layers

Jamin82, 2022-07-24:
**Is your feature request related to a problem? Please describe.**
When selecting only one layer for the top surface I notice that you can see the perimeter extrusion from underneath. This is exaggerated when printing with larger nozzles. shown here circled in red:

[image]

**Describe the solution you'd like**
Being able to select the single perimeter feature for multiple layers has proven to remove this ghosting effect and provide a uniform top surface infill. The current selection allows us to select for either the first layer and / or top surfaces. being able to select the number of top surfaces would make this feature possible

**Describe alternatives you've considered**
as a workaround, I have experimented with using modifiers to change perimeter count on multiple layers and visually it gives the outcome I am after. It is very time-consuming to perform this for every slice

supermerill, 2022-07-24:
> When selecting only one layer for the top surface I notice that you can see the perimeter extrusion from underneath.

Isn't it because of 'over-bridge flow' ? try with it set at 100%

If not, can you give me an example project for me to reproduce?

Jamin82, 2022-07-24:
[example project.zip](https://github.com/supermerill/SuperSlicer/files/9176952/example.project.zip)

this is with 95% infill and 8 solid top layers

Jamin82, 2022-07-25:
[image]

showing part printed with only one perimeter on the topmost layer (left)
Part with only one perimeter on top two layers (right)

supermerill, 2022-07-26:
understood, it's the perimeter-infill encroaching that shows up.

Jamin82, 2022-07-26:
Appreciate you looking into this