supermerill/SuperSlicer #2978: Skirt draft shield consuming massive time and filament

gswatkins, 2022-07-24:
### What happened?

Enabling a draft shield significantly increases the time and filament usage estimates
[image]
.

### Project file & How to reproduce

[Calibration cube.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9176284/Calibration.cube.3mf.zip)
Steps to reproduce:
1. Generate a calibration cube
2. Enable draft shield in Print Settings -> Skirt & Brim
3. Slice

### Version

2.4.58.3

### Operating system

macos 12.4

### Printer model

vcore 3

supermerill, 2022-07-24:
I don't have the same result on my 2.4.58.3

[image]

gswatkins, 2022-07-24:
Interesting. It happens reliably every time for me. Works as expected when I go back to 2.3.57.12. I even tried using presets to rule out any of my specific config settings:
[image]
[image]
Anything else I should try?

supermerill, 2022-07-24:
You can try with the latest nightly, maybe it will fix itself.

gswatkins, 2022-07-25:
The latest macOS arm nightly had failed, but I was able to build from source. I still see the same issue.

ElliottBridge, 2022-07-25:
Maybe it's macos specific?

supermerill, 2022-08-10:
can't reproduce with latest nightly, maybe it corrected itself.

gswatkins, 2022-08-10:
I also tried with the latest nightly and it appears to be fixed. Thanks for your help.

closed 2024-02-11 by supermerill