supermerill/SuperSlicer #2979: Variable Layer Height Not Working in 2.4.58.3

LordFlashmeow, 2022-07-24:
### What happened?

I have "Enable variable layer height feature" enabled in settings, but it's not working when slicing.

[image]
[image]

### Project file & How to reproduce

[Stock-bearing04.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9176951/Stock-bearing04.3mf.zip)

### Version

2.4.58.3

### Operating system

macos 12.4

supermerill, 2022-07-25:
you can fix it for your model by stretching just a little bit to 30.01 height.

inspectionsbybob, 2022-09-19:
NOT FIXED in 2.4.58.4....
Get a "Infill failed" message

closed 2024-02-11 by supermerill