supermerill/SuperSlicer #2982: Re-Slice Now doesn't do anything

inspectionsbybob, 2022-07-25:
### Version
2.4.58.3

### Operating system type + version
Win10

### Behavior
Re-Slice Now doesn't do anything

_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs
All projects

supermerill, 2022-07-25:
what is a re-slice?

inspectionsbybob, 2022-07-26:
Re-slice forces a slice (or re-slice) NOW.  It is often used when the view
has become corrupted or a change was made that did not change the state to
needing to be sliced.

[image: image.png]

Bob Sisson

supermerill, 2022-07-26:
How can I reproduce the problem? I don't see any problem with a random project.
The project file to start from?
what to click then?

also your "image.png" is just a word.

inspectionsbybob, 2022-07-26:
let me try a few different stls again...

catmonger, 2022-08-02:
I can verify that this is an issue, I think this is due to the automatic slices performed by Super compared to Prusa. An example I have encountered is when automatic slicing fails to activate which sometimes happens when changing printer settings Ctrl+R/Reload via GUI will fail as well. It could be that reload has been disactivated because of automatic slicing I may very well be wrong though. I think it should be enabled but then again I am most likely missing the source of this issue.