supermerill/SuperSlicer #2983: Canceling an Object causes wandering at layers

inspectionsbybob, 2022-07-25:
### Version
2.4.58.3

### Operating system type + version
Win10

### Behavior
Had a 5 object model.  Early in a print I could see an object was flawed and canceled that part of the print.

At the next layer increment, the head went for a wander all around the bed for more than 35 seconds, and visited all four corners of the bed and the X-stop twice.   Then it resumed the print just like it was supposed to. Not sure what it was doing, but it wasted a fair amount of time doing it...

I canceled part #5

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs

[Big Eye.zip](https://github.com/supermerill/SuperSlicer/files/9183533/Big.Eye.zip)

supermerill, 2022-07-26:
There is to kind of object "change":
```
; stop printing object Eyeball_1.stl id:0 copy 0
M486 S-1
; printing object Lip_Down_right.stl id:5 copy 0
M486 S5
```
and (for the layer change)
```
; stop printing object Adapter_1.stl id:1 copy 0
M486 S-1
G92 E0
G1 Z1.04
; printing object Eyeball_1.stl id:0 copy 0
M486 S0
```

The printer may use the comment or the M486

I don't see why it does the dance, and even less how I can help you.

inspectionsbybob, 2022-07-26:
I will check with my Printer people to see if it possibly is their issue...

Only thought, is that at the layer change it has to print objects 0-4
before it gets to the code for Object 5, which it should skip or do
something different...  More I think about it, it has to be a printer
issue...

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
***@***.***
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

inspectionsbybob, 2022-10-03:
OK, closing

closed 2022-10-03 by inspectionsbybob