supermerill/SuperSlicer #2984: Start Vase mode at a given Z height

ElliottBridge, 2022-07-25:
So, my problem is I want a print to be partial vase mode, I understand it would be impossible to have it return to normal after vase mode since supports/geometry larger than 1 extrusion would be required, etc. And vase mode doesn't support that (and shouldn't).

What I am suggesting is you can import an object, e.g. a vase with a base built in (not what I need I for, but a good example), you can then set up the settings as you would for a normal print, but select a layer height that you want vase mode to generate at. Then above that point everything will be treated as vase mode. In this case you'd have the base, and then the vase be done in vase mode.

In some instances you might be able to generate a model with the right width for extrusion to 'simulate' partial vase mode, but this would be practically impossible for really complicated geometry, and will always leave seams (and strings if tpu or poor retraction), because it hops from layer to layer. Hence why partial vase mode would be super useful for certain prints!

Thank you for reading this. 🙂

- Elliott

ElliottBridge, 2022-07-26:
> duplicate of #620 or #190

Sorry about that, I didn't see these on my search. The others were listed as hard (they look it). Is my idea solution not much easier to complete?

ElliottBridge, 2022-07-26:
> > duplicate of #620 or #190
> 
> Sorry about that, I didn't see these on my search. The others were listed as hard (they look it). Is my idea solution not much easier to complete?

The others would require a rewrite of how vase mode works. Either for finding and working with any continuous 1 perimeter line (hard to detect for one thing), or requiring retractions to be added for the other (multiple objects, no longer vase mode). My solution would just require it to be above a certain height. The other would be nice, but doesn't seem easily achievable.

supermerill, 2022-07-26:
true

contrapuntal, 2022-08-28:
Hasn't this been implemented in #2998 ? Just set the # of bottom layers

fr0sty1, 2022-11-11:
Adding a concrete use-case for this feature. I have recently started printing "gridfinity" bins and the larger sizes all have a multi-part base:
[image]

The upper portion of the bins have a single perimeter: 
[image]

To print these as a single vase-mode print I would need to specify almost twenty (!!) solid bottom layers before beginning the spiral vase on the outside and that is a tremendous waste of plastic. 

Since a "start vase mode at:" feature does not exist (yet) I have been doing the following as a workaround:

1. Slice model as a spiral vase mode with enough solid bottom layers to complete the floor.
2. Do a Z-cut of the model at the same Z height as the start of the spiral transition
3. Re-slice the model as normal print with 1+ perimeters, infill and solid top/bottom layers
4. Delete everything before the spiral transition from the spiral gcode and everything after the last extrude move of the normal gcode
5. concatenate the files together adding a `G92 E???` to make sure the extruder position is set appropriately at the transition.

spot1984, 2022-11-11:
Better would be an opportunistic vase mode that would enter vase mode for any 2 or more layers containing a single perimeter.  The first layer being a transition from standard to the vase spiral, followed by zero or more standard spiral vase layers, and finally a transition from spiral to standard flat layers if layers begin to have more than a single perimeter or if a flat top is desired on the vase.  The transition layers would ramp filament flow with z height to transition from flat uniform thickness layers to a uniform thickness spiral, basically forming a filament wedge interface between the different types of layers.

Most of this functionality could be built into PerimeterGenerator if the caller could analyze the number of perimeters in each layer.

GlnSevi, 2022-11-29:
I print Airplanes. If i want to print Wings i need a top and buttom solid layer for this I use Simplify3D because u already can add Top and Buttom Solid Layer in Vase Mode. But u can´t switch betwen normal and Vase at any given hight. Would be nice if this works for SuperSlicer then the Slicer has all it need for all my Projects SuperSlicer is my GoTo slicer for all other Projects. Have a nice Day.

blitzaxt, 2023-01-13:
I found a workaround, I stacked two models one on top of the other (using add part) and setup modifiers on the top model to do "External perimeters in vase mode" I also turned off infill/top solid layers. I'm sure this could be improved further but it seems to work OK. I'm attaching a screenshot for reference to the settings I'm talking about.
[image]

GlnSevi, 2023-01-14:
This is nice thank u 😊.