supermerill/SuperSlicer #2985: Feature Request: random positioning of prints on build plate

jaimebpg, 2022-07-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

When printing on wearable surfaces like PEI, arranging the prints always on the center created premature wear on that spot, reducing the overall life expectancy of the surface.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Option to instead of arranging the prints in the center, arrange the prints in a random position.

ElliottBridge, 2022-07-25:
> **Is your feature request related to a problem? Please describe.**
> A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
>
> When printing on wearable surfaces [...]

This sounds great! An even better option would be for it to keep a record of how often areas of the bed are used and either just visualise it and/or arrange according to areas least used

supermerill, 2022-07-26:
> An even better option would be for it to keep a record of how often areas of the bed are used and either just visualise it and/or arrange according to areas least used

At this point, this feature is merging with the ones that want to forbid some built plate area because of damage.

Will-wastelander, 2022-10-11:
This feature sounds great, until you think about the possible overhead and
memory leaks from such feature.