supermerill/SuperSlicer #2986: some supports are generated at very large layer height and below build surface

kevenwyld, 2022-07-26:
### What happened?

When attempting to support a cylindrical object of varying diameters placed on it's side support material is generated with:

1) a much larger layer height than the layer_height or base_layer_height values.
2) is shown below the build surface in the preview.
3) ends up squished or balling up on the nozzle from what appears to be overextrusion.

[image]

In the picture below you can see the support material is much much thicker than the rest of the part but only for layers between the bottom and top surface.  I assume the blue "overhang perimeter" which is below the build surface is also a support which is getting distorted somehow because of the size (very small) of the part.

[image]

### Project file & How to reproduce

I'm not sure what to put here other than the 3mf file which demonstrates the problem.  Just slicing the part seems to reproduce it, I've attempted to adjust many settings for layer height and supports without much luck.

[losi_xxxnt_axel_cv_example_for_github.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9186132/losi_xxxnt_axel_cv_example_for_github.3mf.zip)

### Version

2.3.57

### Operating system

archlinux (appimage)

### Printer model

RailCore II 300ZL

kevenwyld, 2022-07-26:
More info on this, I turned off supports and that blue line below the build surface is still there.  So I guess it's not related to supports at all...  Maybe just a really tall overhang perimeter for some reason.

neophyl, 2022-07-26:
Not a bug.  Normal and expected operation.
The thick blue sections are the default bridging/overhang flow settings.  Basically any extrusion not squished onto the layer below is assumed to have a cross section equal to the nozzle diameter.  This means it models the layer as being 0.4mm thick (or whatever your nozzle is set to).  It was that way forever in both PS and SuSi.  Many many github posts about it both here and on the PS github.  

In the 2.4x versions of Prusa Slicer they added a new tickbox to change that fundamental assumption after a lot of posts and much time.  
Super Slicer does it bit differently.  
For example in 2.4 under Print Settings>Width and Flow > Flow theres a dropdown for Bridge flow baseline.  Change that from Nozzle Diameter to layer height (or just tick the simulate prusa no thick bridge box)

However as you are using 2.3 then that's not available to you.  As the 2.3 builds are based on PS 2.3.x which didn't have the option either and always assumes a thick bridge.  If you want to stop that in SuSi 2.3 then the only workaround is to set your thresholds for overhangs on Perimeter & Shell > Overhangs> Bridge Flow to something high, like 200%.  Or higher, depends on your model.

kevenwyld, 2022-07-26:
@neophyl thanks for the explanation.  I did search but since I thought the problem was supports at first I couldn't find much.  This makes a lot more sense.

I will probably try out the 2.4 version as it sounds like it has many other improvements as well and I think that particular feature will help a lot with these very small dense parts.

closed 2022-07-26 by kevenwyld

supermerill, 2022-07-26:
> If you want to stop that in SuSi 2.3 then the only workaround is to set your thresholds for overhangs on Perimeter & Shell > Overhangs> Bridge Flow to something high, like 200%. Or higher, depends on your model.

You can just reduce your bridge flow, no?

kevenwyld, 2022-07-26:
@supermerill sortof...  Reducing bridge flow seems to not fix the weird layer height shown in the preview.  The workaround suggested above just prevents the creation of bridges using "bridge flow".

[image]

[image]

The center of the layer for those "fatter" bridges seems to be nearly exactly the height of the build surface according to the preview, though the gcode shows a reasonable Z height (in this case `0.22` which seems right for a `0.14` base layer height + some kind of offset to make room for the extra plastic) but since the flow is so high I guess it gets rendered in the preview as below the surface.

Anyway 2.4 seems to fix basically all these problems and offers more control over where to apply thicker bridges so I'm not sure there's anything to really do here.