supermerill/SuperSlicer #2989: SuperSlicer New version link goes to wrong place.

barry99705, 2022-07-26:
### What happened?

Opened up superslicer and get the new version available popup at the bottom of the screen.
Click the link and it goes to prusaslicer's page.
[image]

### Project file & How to reproduce

Opened up superslicer and get the new version available popup at the bottom of the screen.
Click the link and it goes to prusaslicer's page.

### Version

2.4.58.2

### Operating system

Windows 11

supermerill, 2022-07-26:
duplicate of #2954

Stooovie, 2022-10-30:
Not fixed as of 2.4.58.3. Update tells me there's PrusaSlicer update available, and then opens a PrusaSlicer download page.

closed 2024-02-11 by supermerill