supermerill/SuperSlicer #2995: Seam gap does not seem to be working

jmccaull, 2022-07-28:
### What happened?

I'm not sure if this is a bug or not, but no matter what seam gap I set I seem to get the same results. The gcode preview shows a distinct gap when I put a .5mm gap but the print has none and looks identical to 0 seam gap.
I don't actually want a gap, I was testing this trying to get the seam filled in more, as it looks like a little indent but noticed the setting wasn't making any difference in the print. I've tried with and without all types of wipe, but I can't get my seams to look any better. Any suggestions if this is working as intended would be welcome, or maybe even the ability to set a negative seam gap and overlap the ends a little more?
Gcode preview with the gap:
[image]
Resulting print:
[image]

Gcode without gap:
[image]

resulting print:
[image]

I'm using Klipper if that matters and the results are the same with/without pressure advance configured.

Thanks!

### Project file & How to reproduce

[bencyStackWithGap.zip](https://github.com/supermerill/SuperSlicer/files/9204353/bencyStackWithGap.zip)

### Version

2.4.58.3

### Operating system

Windows 10

### Printer model

Sidewinder X1

supermerill, 2022-07-28:
Seam gap is here to create a gap to avoid over-extrusion on pressure release at the end of the extrusion.
If you want the opposite there is "extra wipe for external perimeter" and "wipe_inside"

> maybe even the ability to set a negative seam gap and overlap the ends a little more?

Maybe you underextrude a little bit. You can push the filament flow multiplier just a touch.

Also, there is a "vase external perimeter" in the perimeter tab, in the "external perimeter first". I created this to test an idea, but didn't find it any better. You can try it and report if it's good or not at all.

tekhedd, 2022-11-30:
I've been trying to optimize seams today, and I'm seeing the same thing.

Thought maybe I was imagining it so I set seam gap to 3mm. No change whatsoever in the print, although the generated g-code shows 3mm of motion with no E motion and octoprint displays those mm as non-print moves. It's unnatural. Maybe this is a klipper bug? Since it does look like SS is generating the seam gap in the g-code, it might not be a bug in the slicer. I appreciate that it might be oozing a bit, but for 3mm? Yeah I really doubt that too.

This is with 2.4.58.5, AppImage/gtk, Klipper.

berimbolero, 2023-02-11:
I have the same thing going on with this seam gap feature. No difference in print between 8mm and 0.4mm gap.

CottonRunt, 2023-02-12:
Same, not sure what might be causing it but seam gap settings no longer seem to have any effect on prints.

AnthonyBerduck, 2023-02-19:
Same problem here, got big holes in gcodes. Tweaking "Extra wipe for external perimeter" and "wipe_inside" don't change anything.

SpiggieCode, 2024-09-04:
Ill go ahead complain here too, seam gap doesnt appear to be working for me either. 0% and 10% both had no difference.