supermerill/SuperSlicer #2996: Access violation has occured

lanceofwhichwedream, 2022-07-28:
### What happened?

I have a printer, print and filament profile created for printing Voron 2.4r2 parts based on their recommended settings (as well as the settings for the filament I've found to work.). Attempting to slice this results in the error listed above with no additional information.

### Project file & How to reproduce

[Voron24r2-2.zip](https://github.com/supermerill/SuperSlicer/files/9205013/Voron24r2-2.zip)

open project, attempt to slice.

### Version

Version 2.4.3

### Operating system

Windows 11 & Ubuntu Jammy

### Printer model

Prusa MK3S with LGX and Mosquito

supermerill, 2022-07-28:
can you test this nightly version https://github.com/supermerill/SuperSlicer/suites/7554377417/artifacts/311914789 ?
sadly it's a random bugand so very difficult to pnpoint, fix and even to verify if it's fixed.

lanceofwhichwedream, 2022-07-29:
> can you test this nightly version https://github.com/supermerill/SuperSlicer/suites/7554377417/artifacts/311914789 ? sadly it's a random bug and so very difficult to pinpoint, fix and even to verify if it's fixed.

Sure thing, I had a feeling it wouldn't be an easy issue to pinpoint. I think there's a specific piece causing this. I don't know the exact piece, but I'm narrowing it down currently.

lanceofwhichwedream, 2022-07-29:
Tested, still crashed. I couldn't find a crash_log.txt but I did find the part that's causing this. [a]_keystone_blank_insert.stl

supermerill, 2022-07-30:
> but I did find the part that's causing this.

what part is it?
I think you tried to send it via an email, but that doesn't work. If possible, It's better to export your project just before clicking on the dangerous button (slice?), if possible.

Also, to be sure it's not coming from gui (unlikely), you can slice your project with this command:
`superslicer-console -s "C:\Users\lance\3D Objects\_keystone_blank_insert.3mf"` and check if it crash as expected.

lanceofwhichwedream, 2022-07-30:
I tried slicing with the console command but this is my only output, 
`C:\Users\Lance Zeligman\Downloads\SuperSlicer_2.4.58.3_win64_220714\SuperSlicer_2.4.58.3_win64_220714>superslicer_console.exe -s "keystone.3mf"
[2022-07-30 13:44:56.166927] [0x00004678] [trace]   Initializing StaticPrintConfigs
Detected technology ☺
[keystone.zip](https://github.com/supermerill/SuperSlicer/files/9227032/keystone.zip)

C:\Users\Lance Zeligman\Downloads\SuperSlicer_2.4.58.3_win64_220714\SuperSlicer_2.4.58.3_win64_220714>`

No gcode file was produced. Additionally I saved the project and I'm reuploading it. The only model visible is the trouble making one. I'm uploading it as a zip, you'll find the project file inside.

lanceofwhichwedream, 2022-07-30:
[keystone.zip](https://github.com/supermerill/SuperSlicer/files/9227033/keystone.zip)

supermerill, 2022-08-03:
tahnks for the project. Found the problem. Fix in next nightly.

closed 2024-02-11 by supermerill