supermerill/SuperSlicer #2999: Change bed temperature and change extruder temperature at layer number to be set

ShaneF1, 2022-07-28:
When printing ABS I have found that leaving the bed temperature high for several layers rather than one before temperature reduction reduces curling. This allows the first few layers to remain hot while built upon, then allows them to cool together as a block and provide a stiffer base to resist curling forces from upper layers. The current change at layer one capability causes immediate curling forces on the weak hot first layer.

The interface could have layer selection in the filament bed "other layers" temperature setting, similar to the "full fan speed at layer" selection in filament cooling. While I have no current need, having the capability to select the layer at which extruder temps change might also be useful. Having a default of one for these setting leaves the current behaviour unchanged yet provides additional function for those who may benefit.

This can be done now by leaving the filament bed first layer temps unchanged in the slicer and manually editing the gcode file to insert M140 commands where desired. This could also be done using post processing scripts. The first option is error prone, the second requires coding skills and a custom configured slicer environment.

[image]

supermerill, 2022-07-28:
How to do that currently:
in filament -> notes-Custom variables:
```
temp_layer=10
bed_temp_layer=10
```
in printer->custom gcode->Start g-code
```
default_int(temp_layer,0)
default_int(bed_temp_layer,0)
```
in printer->custom gcode -> After layer change
```
{if temp_layer >= layer_num}M104 S{((first_layer_temperature *(temp_layer - layer_num) +temperature * layer_num)/temp_layer)+extruder_temperature_offset}{endif}
{if bed_temp_layer >= layer_num}M140 S{((first_layer_bed_temperature *(bed_temp_layer - layer_num) +bed_temperature * layer_num)/temp_layer)}{endif}
```
As in the current version, the temperature change is set after the "after layer change", this needs to be added to the "between extrusion role custom gcode"
```
{if temp_layer >= layer_num && layer_num == 2}M104 S{((first_layer_temperature *(temp_layer - layer_num) +temperature * layer_num)/temp_layer)+extruder_temperature_offset}{endif}
{if bed_temp_layer >= layer_num  && layer_num == 2}M140 S{((first_layer_bed_temperature *(bed_temp_layer - layer_num) +bed_temperature * layer_num)/temp_layer)}{endif}
```

ShaneF1, 2022-07-29:
Thank you.

The only downsides are that every filament now has to have those variables defined or slicing breaks when using a printer with the code amendments, and slicer behaviour is now different from what the GUI indicates. This not a problem if the user is well engaged, but a user coming back to the hobby after an extended break (as I am) would be quite confused.

This code does almost perfectly what I was hoping to achieve. It has given me a template for how to do a step change rather than gradual change.