supermerill/SuperSlicer #3001: 3dconnexion SPACE MOUSE not working anymore

cristianku, 2022-07-28:
### What happened?

3dconnexion SPACE MOUSE not working anymore. it was working fine with the previous Mac version( non arm ), on Mac mini m1.

now I have installed the last version ( ARM this time ) on Mac mini M1, and the 3d mouse 3dconnexion has stopped working.

its working fine with all other applications. for example with Fusion 360.

I am talking about this 
https://3dconnexion.com/us/product/spacemouse-wireless/

### Project file & How to reproduce

3dconnexion SPACE MOUSE not working anymore. it was working fine with the previous Mac version( non arm ), on Mac mini m1.

now I have installed the last version ( ARM this time ) on Mac mini M1, and the 3d mouse 3dconnexion has stopped working.

its working fine with all other applications. for example with Fusion 360.

I am talking about this 
https://3dconnexion.com/us/product/spacemouse-wireless/

### Version

2.4.58.3

### Operating system

Mac OS  12.4 - M1

### Printer model

Voron 2.4

ntrp, 2022-10-19:
I have the same issue, any way we can provide debug info to help fix the problem?

ntrp, 2022-10-19:
I found the logs, I guess it can be of help:
`3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 0x0005): tried: '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (code signature in <4CF19714-B8C0-395F-A3E9-DFE0D14D6DD5> '/Library/Frameworks/3DconnexionClient.framework/Versions/A/3DconnexionClient' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs), '/System/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/Library/Frameworks/3DconnexionClient.framework/Versions/A/3DconnexionClient' (code signature in <4CF19714-B8C0-395F-A3E9-DFE0D14D6DD5> '/Library/Frameworks/3DconnexionClient.framework/Versions/A/3DconnexionClient' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs), '/System/Library/Frameworks/3DconnexionClient.framework/Versions/A/3DconnexionClient' (no such file)`

SS version: 2.4.58.5
MacOS: macOS 12.6 (21G115)
3DxMacCore Version: 1.2.4.307 (307)
3DxWare Version: 10.7.4 (3493)

denis-itskovich, 2022-10-28:
I have the same issue on Windows 11, same SuperSlicer version (2.4.58.5). 
Might be related to #298

ZyberSE, 2023-03-14:
Same issue here too, MacBook Pro M1 MAX (macOS 13.2.1) SuperSlicer 2.4.58.5

ruok5, 2023-03-22:
If you're on an Apple Silicon Mac, it's worth trying the non-ARM binary. I just installed 2.5.59.2 (pre-release) on my M1 (13.2.1), and my Space Mouse **_now works fine_**. I suspect this has to do with the 3dconnexion software (10.7.4) running under Rosetta 2, and forcing SS to do the same is working as a short-term solution. Their [software manual](https://3dconnexion.com/us/wp-content/uploads/sites/27/2023/01/Release_Notes-Mac-20230109.pdf) states that the drivers were updated to support Apple Silicon as of 10.7.0 (Jan 2022), but who knows _to what extent_ other than compatibility.